atomvm/AtomVM #887: Implement an MVP of the OTP `ets` interface

fadushin, 2023-10-23:
There are cases where AtomVM could benefit from `ets` tables.

This task would involve implementing a “minimal viable product” for `ets` in AtomVM.  In essence, the minimal implementation should support the ets (unordered) `set` type, with operations for insertion, lookup, and deletion.  Support for sharing tables across processes with support for read-concurrency is desirable, but not strictly needed for the MVP.

Implemented functions of `ets` (with caveats)would include (but may not necessarily be limited to) the following:

* `ets:new/2` with support for the `set` type, `key_pos`, `named_table`, `private`, `public`, and `protected` access, and `read_concrrency` (opportunistic)
* `ets:member/2`
* `ets:insert/2` and `ets:insert_new/2`
* `ets:lookup/2` and `ets:lookup_element/3`
* `ets:delete/2` and `ets:delete_all_objects/1`
* `ets:tab_to_list/1`
* `ets:whereis/1`

ceedon, 2024-04-25:
I'm unable to get gleam/mist running due to a dependency on ets:lookup_element/3 .
Does anyone intend to push the rest of this work forward?

I fully understand the unfortunate circumstances around it being partially completed, so forgive me for asking.
I'm trying to understand if I need to start learning C to help do so.

uwiger, 2025-07-08:
FWIW, I have an old erlang implementation of a match specification interpreter.
https://github.com/gebi/jungerl/blob/master/lib/rdbms/src/rdbms_ms.erl

Feel free to use it as a stop-gap solution, if it helps.
As it's almost 20 years old, it might need some updating. :)