atomvm/AtomVM #894: Return {error, Reason} from `gpio` operations in error conditions

fadushin, 2023-10-25:
The AtomVM `gpio` interface returns bare `error` atoms in error conditions, which can make it difficult to diagnose why an error occurred.

We need to change the return type of these operations from `error` to `{error, Reason :: term()}`, where `Reason` is an appropriate term that can be used for logging or even for a program to take action.

The following functions need to be modified:

* `gpio:close/0`
* `gpio:stop/0`
* `gpio:set_direction/3`
* `gpio:set_level/3`
* `gpio:set_int/3`
* `gpio:remove_int/2`
* `gpio:set_pin_mode/2`
* `gpio:set_pull_mode/2`
* `gpio:hold_en/1`
* `gpio:hold_dis/1`
* `gpio:digital_write/2`
* `gpio:attach_interrupt/2`
* `gpio:detach_interrupt/1`

pguyot, 2023-10-27:
We definitely need to align semantics of gpio module across platforms.
We may want to fail with badargs when arguments are inappropriate instead of returning error atom as we currently do.

fadushin, 2023-10-27:
I agree on badarg errors as well.  Let’s take a look at patterns in OTP for guidance on expected behavior.

UncleGrumpy, 2023-10-27:
In almost all cases `badarg` does make sense. With some exceptions, `gpio:stop/0` and `gpio:close/1` should probably `RAISE_ERROR(NOPROC_ATOM)` if no gpio driver process is currently running, that might also be the most appropriate response to attempting to remove an interrupt that isn't set. I'm not sure what the return for setting an interrupt on a pin that is already attached should be, the argument is still valid.