atomvm/AtomVM #895: terminal output question/issue.

dexmagic, 2023-10-25:
So I'm trying to understand how AtomVM is outputting things to the terminal.   When running on the Mac, I don't see output that I would expect when I modify nif_erlang_display_1

I updated `./src/libAtomVM/nifs.c` to this:

```
static term nif_erlang_display_1(Context *ctx, int argc, term argv[])
{
    UNUSED(ctx); 
    UNUSED(argc);

    printf("nif_erlang_display_1\n"); // ADDED THIS LINE EXPECTING TO SEE IT DURING RUNTMIE.

    term_display(stdout, argv[0], ctx);
    printf("\n");
    
    return TRUE_ATOM;
}
```

And I wrote the following Erlang progam:

```
% more run.erl
-module(run).
-compile(export_all).

start() ->
        io:format("Hello~nWorld~n").
```

After compiling it, converting it to an avm I run it:

```
% AtomVM ./run.avm ./atomvmlib.avm
Hello
World
Return value: ok
```

I would expect to see 

```
% AtomVM ./run.avm ./atomvmlib.avm
nif_erlang_display_1
Hello
World
Return value: ok
```

is nif_erlang_display_1  not being called or is there another reason output from printf would be supporessed?

Here's the relevant Erlang code from the library. From `./libs/estdlib/src/io.erl`:

```
-spec format(Format :: string()) -> string().
format(Format) when is_list(Format) ->
format(Format) when is_list(Format) ->
    format(Format, []).

%%-----------------------------------------------------------------------------
%% @param   Format format string
%% @param   Args format argument
%% @returns string
%% @doc     Format string and data to console.
%%          See io_lib:format/2 for information about
%%          formatting capabilities.
%% @end
%%-----------------------------------------------------------------------------
-spec format(Format :: string(), Args :: list()) -> string().
format(Format, Args) when is_list(Format) andalso is_list(Args) ->
    Msg =
        try
            io_lib:format(Format, Args)
        catch
            _:_ ->
                io_lib:format("Bad format!  Format: ~p Args: ~p~n", [Format, Args])
        end,
    put_chars(Msg).

```

fadushin, 2023-10-25:
`io:format` does not use the `term_display1 nif.  Try `erlang:display(Term)` instead.

dexmagic, 2023-10-25:
Good to know. Thanks. 

So follow up question then: What NIF does io:format resolve to? 

Would be good to know. I have lots of Erlang code that already uses io:format.

dexmagic, 2023-10-25:
Looks like this is related?

```
%% @private 
interleave([LastToken], _Instr, [], Accum) ->
    lists:reverse([LastToken | Accum]);
interleave([Token | Tokens], [Formatter | Instr], [Arg | Args], Accum) ->
    interleave(Tokens, Instr, Args, [Formatter(Arg), Token | Accum]).
```

fadushin, 2023-10-26:
> Good to know. Thanks.
> 
> So follow up question then: What NIF does io:format resolve to?
> 
> Would be good to know. I have lots of Erlang code that already uses io:format.

https://github.com/atomvm/AtomVM/blob/25c09a0527fc45f8dda778f803e07751d4f96a6a/src/libAtomVM/nifs.c#L3814

fadushin, 2023-10-26:
> Looks like this is related?
> 
> ```
> %% @private
> interleave([LastToken], _Instr, [], Accum) ->
>     lists:reverse([LastToken | Accum]);
> interleave([Token | Tokens], [Formatter | Instr], [Arg | Args], Accum) ->
>     interleave(Tokens, Instr, Args, [Formatter(Arg), Token | Accum]).
> ```
> 
> Apologies for all the questions ... I'm just loosing the code-trail for io:format.

What it is it that you are trying to sort out here?

bettio, 2024-03-05:
There are no updates on this issue, closing it.

closed 2024-03-05 by bettio