atomvm/AtomVM #915: AtomVM crashes on STM32 with no console messages

vijay-packetp, 2023-11-01:
When using AtomVM on an STM32 platform, we are experiencing crashes with no console messages displayed. The issue seems to be related to a crash occurring at line 85 in the sys.c file in the AtomVM repository: https://github.com/atomvm/AtomVM/blob/3ebab2b7acea3883a591f740b2c374f498fe50ab/src/platforms/stm32/src/lib/sys.c#L85. This crash is leading to a hard fault handler in the main.c file of the STM32 platform, specifically within the sys_lock_pin() function.

glb->platform_data was not allocated,

vijay-packetp, 2023-11-01:
The issue resolved by allocating memory for glb->platform_data in the sys_init_platform function. Specifically, added the following code to ensure that memory is properly allocated and initialized for the STM32 platform data:
```
struct STM32PlatformData *platform = malloc(sizeof(struct STM32PlatformData));
list_init(&platform->locked_pins);
glb->platform_data = platform;
```

UncleGrumpy, 2023-11-01:
Thank you for this report! It there is definitely some code in `sys_init_platform` that somehow didn't get merged when this driver was updated. I will investigate and should have a fix in short order.

UncleGrumpy, 2023-11-01:
PR #918 has been submitted with the correction. Thank you again for opening an issue.

PR #918 (2023-11-01): Fix STM32 GPIO driver Issue #915
During the PR review and snd subsequent bug fixes the code required for `sys_platform_init` was inadvertantly dropped.

Closes #915

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit f3b419c8 (2023-11-01): Fix STM32 GPIO dirver Issue #915

During the PR review and snd subsequent bug fixes the code required for
`sys_platform_init` was inadvertantly dropped.

Closes #915
closed 2023-11-01 by fadushin