atomvm/AtomVM #932: Linker script created by libopencm3 for STM32 causes memory bugs

UncleGrumpy, 2023-11-07:
The linker script generated by libopencm3 for the STM32 platform does not define the end of the `stack` or `heap`. This leads to memory allocation failures on the platform to cause a hard fault rather than return memory related errors. I have been working on better fault handling by decoding the fault registers and noticed that failed malloc and related calls always return a hard fault error rather than a memory error, and the VM is never able to abort gracefully.

The reason is that without knowing where the end of the stack and heap are there is nothing to stop `malloc` from attempting to allocate a protected hardware register as ram, which will always trigger a hard fault. The linker script generator is GPL-3, so rather than modify the generator script (as discussed in PR #897), a new way of generating the linker script must be developed.

I believe it would be easier to do this in Erlang, than attempt to create a similar python script and device data file from scratch. The most time consuming part will be creating a data set of all the possible STM32 chips that AtomVM can support.

UncleGrumpy, 2023-11-11:
I can confirm the failing sodoku tests on stm32f429zi are indeed a memory related bug in the linker script generated by libopencm3. It does not define an end of stack - which leads to the hard faults. By manually adding the a line to the generated linker that defines the `_estack` all of the atomvm_benchmark test now pass:
```

    ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ##
      ## ##      ##    ##     ## ###   ### ##     ## ###   ###
     ##   ##     ##    ##     ## #### #### ##     ## #### ####
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ##
    #########    ##    ##     ## ##     ##  ##   ##  ##     ##
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ##
    ##     ##    ##     #######  ##     ##    ###    ##     ##

    ###########################################################

INFO [52] AtomVM: Starting AtomVM revision 0.6.0-alpha.2+git.f5bb564b
DEBUG [59] AtomVM: Using usart mapped at register 0x40004800 for stdout/stderr.
DEBUG [67] AtomVM: Maximum application size: 1048576
INFO [73] AtomVM: Booting file mapped at: 0x8080000, size: 1792
INFO [80] AtomVM: Starting: benchmark.beam...

---
Running tests:
pingpong_speed_test: 5981000
prime_speed_test: 17554000
prng_test: 379000
pi_test: 25922000
sudoku_solution_test: 8073000
sudoku_puzzle_test: 305185000
INFO [363201] AtomVM: Exited with return: ok
INFO [363206] AtomVM: AtomVM application terminated.  Going to sleep forever ...
```

bettio, 2023-11-15:
I think an issue here on GitHub should be opened against libopencm3 on both sides.
Once opened, please reference the libopencm3 issue here.

UncleGrumpy, 2023-11-15:
I searched through the "Closed" issues in the `libopencm3` repo and found a similar issue, that suggested a fix by exporting boundaries in the linker script.  The issue and associated PR were both closed but ended up having a link to a page in the [libopencm3 wiki](https://github.com/libopencm3/libopencm3/wiki/Using-malloc-with-libopencm3) that explains a work around for safely using malloc. I have implemented that fix in PR #946

commit 2eeb24f2 (2023-11-17): Merge pull request #946 from UncleGrumpy/stm32_heap_fix

Fix STM32 hard fault when using malloc

Fixes a bug that causes a hard fault when suplimental sram chips are accessed on
devices that have more than one bank of ram. This redefinition of `_sbrk_r` is
the recommended solution on the
[libopencm3 wiki](https://github.com/libopencm3/libopencm3/wiki/Using-malloc-with-libopencm3).

Closes #932

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit d73922d1 (2023-11-17): Fix STM32 hard fault when using malloc

Fixes a bug that causes a hard fault when suplimental sram chips are access on
devices that have more than one bank of ram. This redefinition of `_sbrk_r` is
the recommended solution on the [libopencm3 wiki]
(https://github.com/libopencm3/libopencm3/wiki/Using-malloc-with-libopencm3).

Closes #932
closed 2023-11-17 by bettio