atomvm/AtomVM #942: License conflict building AtomVM with libopencm3?

UncleGrumpy, 2023-11-13:
The discussion around modifying the linker script source file in PR #897 has drawn my attention to the fact that libopencm3 is GPL-3 licensed, and has no special clause allowing for the use of the linked libraries in projects with more permissive licenses. A discussion was started as few years ago in the libopencm3 project about adding such a clause, but to this date no change has been made. This leaves it up for debate whether using libopencm3 effectively forces the entire build to be GPL-3.

See the discussion on the libopencm3 wiki:

https://github.com/libopencm3/libopencm3/wiki/Licensing

bettio, 2023-11-14:
This applies only to the STM32 port until we use libopencm3.

As far as I know, from a practical point of view this is an issue when trying to distribute a closed source (or any GPL not compliant) version of the STM32 native firmware.

This is not an issue for any proprietary application running on the VM on STM32.

I think we should close this issue since I don't see any actual conflict, however feel free to open an issue about adopting either NuttX or Zephyr that allows proprietary STM32 customized base firmwares.

UncleGrumpy, 2023-11-15:
We should probably mention this somewhere in the documentation for the time being. With the general lack of maintenance, and the recently discovered bug #932 with some libopencm3 device builds (which requires modifying the LGPL-3 linker script template to fix) migrating to a different base for stm32 seems advantageous.

closed 2023-11-15 by UncleGrumpy