atomvm/AtomVM #992: Unsupported line_ref tag: 28 followed by sigabrt

rtATvw, 2023-12-18:
The dependency from which the unsupported line_ref seems to be from is: travelping/gen_netlink 
(with calls to lager replaced with io:format to avoid 9 arity limit elsewhere)
Erlang version OTP/25

An abort is triggered in opcodeswitch.h:4438 after the following error:

Unsupported line_ref tag: 28

Contents of module.c / parse_line_refs / *data which triggered warning

```
12 09 AE 22 09 6F 09 AA  09 D6 29 BB 49 1B 49 22  |  ...".o....).I.I" 
49 1F 49 25 49 58 49 4C  49 49 49 46 49 43 49 3A  |  I.I%IXILIIIFICI: 
49 37 49 2E 49 2B 49 28  49 68 49 72 49 6C 49 79  |  I7I.I+I(IhIrIlIy 
49 87 49 91 49 8B 49 98  49 A6 49 AD 49 AA 49 B4  |  I.I.I.I.I.I.I.I. 
49 C1 49 BE 49 BB 49 B8  49 C8 12 49 A0 22 49 EB  |  I.I.I.I.I..I."I. 
49 FC 69 06 69 03 69 00  69 0D 69 11 69 24 69 4D  |  I.i.i.i.i.i.i$iM 
12 49 80 22 69 58 69 6C  69 7A 69 84 69 81 69 7E  |  .I."iXilizi.i.i~ 
12 49 18 22 69 9A 69 A1  69 9E 69 A8 69 B5 69 B2  |  .I."i.i.i.i.i.i. 
69 AF 69 AC 69 BC 69 DF  12 49 B8 22 69 E6 69 E3  |  i.i.i.i..I."i.i. 
89 1A 12 49 BA 22 89 2D  89 3D 89 7E 89 9E 89 C7  |  ...I.".-.=.~.... 
A9 1E A9 37 A9 34 A9 2E  A9 25 A9 22 A9 53 A9 85  |  ...7.4...%.".S.. 
A9 9B A9 95 A9 92 A9 8C  A9 89 A9 A2 A9 C7 A9 A9  |  ................ 
A9 A6 A9 F1 A9 D9 C9 4A  C9 5B 12 49 B3 22 C9 7B  |  .......J.[.I.".{ 
C9 7F C9 7C C9 86 C9 91  C9 95 C9 9F C9 DE C9 F5  |  ...|............ 
E9 17 E9 47 E9 55 E9 5C  E9 63 E9 6A E9 71 E9 93  |  ...G.U.\.c.j.q.. 
E9 A3 E9 A0 E9 AA E9 BE  E9 C5 E9 CF 19 08 1C 19  |  ................ 
08 3E 19 08 6B 19 08 72  19 08 7F 19 08 AE 19 08  |  .>..k..r........ 
BF 19 08 C9 19 08 E7 19  08 F8 19 09 09 19 09 0D  |  ................ 
19 09 14 19 09 22 19 09  41 19 09 52 19 09 5F 19  |  ....."..A..R.._. 
09 6B 19 09 72 19 09 9A  19 09 A7 19 09 B7 19 09  |  .k..r........... 
FA 19 0A 04 19 0A 0E 19  0A 12 19 0A 19 19 0A 20  |  ...............  
19 0A 1D 19 0A 2F 19 0A  36 19 0A 41 19 0A 51 19  |  ...../..6..A..Q. 
0A 8A 19 0A AF 19 0A D2  19 0A E8 19 0A E5 19 0A  |  ................ 
D6 19 0A EF 19 0A FD 19  0B 0B 19 0B 16 19 0B 38  |  ...............8 
19 0B 60 19 0B 48 19 0B  45 19 0B 67 19 0B A2 19  |  ..`..H..E..g.... 
0B CE 19 0C 78 19 0C 94  19 0C B3 19 0D 02 19 0D  |  ....x........... 
13 12 49 29 49 1B 22 19  0D 1A 19 0D 17 19 0D 3E  |  ..I)I."........> 
19 0D 50 19 0D 1D 19 0D  20 19 0D 26 19 0D 44 19  |  ..P..... ..&..D. 
0D 41 19 0D 3B 19 0D 23  19 0D 32 19 0D 2F 19 0D  |  .A..;..#..2../.. 
61 19 0D 6B 19 0D 73 19  0D 82 19 0D 8C 19 0D 94  |  a..k..s......... 
19 0D A3 19 0D AA 19 0D  A7 19 0D B2 19 0D C7 12  |  ................ 
49 45 49 27 22 19 0D EB  12 49 2B 22 19 0D FD 19  |  IEI'"....I+".... 
0E 01 19 0E 07 19 0E 04  19 0E 0F 19 0E 13 19 0E  |  ................ 
27 19 0E 50 12 49 24 22  19 0E 5C 12 49 2D 22 19  |  '..P.I$"..\.I-". 
0E 71 19 0E 80 19 0E 87  19 0E 84 19 0E 8A 12 49  |  .q.............I 
08 49 09 22 19 0E A1 19  0E A8 19 0E A5 19 0E B0  |  .I."............ 
19 0E C5 19 0E E9 19 0F  26 19 0F 39 12 49 33 22  |  ........&..9.I3" 
19 0F 4B 12 49 31 22 19  0F 8D 19 0F AD 19 0F D6  |  ..K.I1"......... 
19 10 05 19 10 19 19 10  2D 19 10 46 19 10 64 19  |  ........-..F..d. 
10 97 19 10 B6 19 10 DC  19 10 EE 19 11 07 19 11  |  ................ 
61 19 11 73 12 49 59 22  19 11 98 19 11 9C 19 11  |  a..s.IY"........ 
99 19 11 A4 19 11 B1 19  11 C1 19 11 DC 19 12 01  |  ................ 
19 12 06 19 12 0A 19 12  1B 19 12 20 19 12 24 19  |  ........... ..$. 
12 2B 19 12 3B 19 12 41  19 12 72 19 12 81 19 12  |  .+..;..A..r..... 
88 19 12 90 19 12 97 19  12 9F 19 12 C2 19 12 D2  |  ................ 
19 12 CF 19 12 DA 19 12  EF 19 12 F6 19 13 01 19  |  ................ 
13 4E 19 13 71 19 13 9F  19 13 A6 19 13 B4 19 13  |  .N..q........... 
E3 19 13 F5 19 13 FF 19  14 1E 19 14 30 19 14 42  |  ............0..B 
19 14 46 19 14 4E 19 14  5D 19 14 7D 19 14 8F 19  |  ..F..N..]..}.... 
14 9C 19 14 A8 19 14 B0  12 49 3A 22 19 14 D9 19  |  .........I:".... 
14 E6 19 14 F7 19 15 3B  19 15 45 19 15 50 19 15  |  .......;..E..P.. 
54 19 15 5C 19 15 60 19  15 72 19 15 63 19 15 7A  |  T..\..`..r..c..z 
19 15 86 19 15 96 19 15  D0 19 15 F6 19 16 19 19  |  ................ 
16 2C 19 16 2F 19 16 1D  19 16 37 19 16 46 19 16  |  .,../.....7..F.. 
55 19 16 61 19 16 73 19  16 84 12 29 81 29 86 29  |  U..a..s....).).) 
88 29 8E 29 A7 29 B0 29  B3 29 BC 29 BF 29 C1 29  |  .).).).).).).).) 
C2 29 C4 29 C5 29 C7 29  E0 29 E5 29 E6 29 E8 29  |  .).).).).).).).) 
EC 29 ED 29 EF 29 F1 29  F3 29 F5 29 F6 29 F8 29  |  .).).).).).).).) 
FB 29 FE 49 00 49 02 49  07 49 0C 49 11 49 12 49  |  .).I.I.I.I.I.I.I 
17 49 1A 49 2A 49 2E 49  2F 49 30 49 3F 49 40 49  |  .I.I*I.I/I0I?I@I 
46 49 47 49 49 49 4A 49  4D 49 50 49 4E 49 52 49  |  FIGIIIJIMIPINIRI 
54 49 5B 49 5C 49 5E 49  5F 49 61 49 64 49 62 49  |  TI[I\I^I_IaIdIbI 
66 49 67 49 69 49 6B 49  6D 49 71 49 72 49 74 49  |  fIgIiIkImIqIrItI 
75 49 A1 49 A7 49 AC 49  BC 49 BD 49 C1 49 CC 49  |  uI.I.I.I.I.I.I.I 
CD 49 CF 49 D0 49 D6 49  D7 49 D9 49 DF 49 E0 49  |  .I.I.I.I.I.I.I.I 
DD 49 E5 49 E7 49 EA 49  ED 49 F2 49 F4 49 F8 49  |  .I.I.I.I.I.I.I.I 
FB 49 FC 49 F9 49 FF 69  02 69 6D 69 6E 69 6F 69  |  .I.I.I.i.iminioi 
72 69 73 69 74 69 29 69  2A 69 2B 69 2C 69 2D 69  |  risiti)i*i+i,i-i 
2E 69 31 69 33 69 34 69  35 69 36 69 32 69 44 69  |  .i1i3i4i5i6i2iDi 
45 69 46 69 0E 69 0F 69  11 69 13 69 14 69 3B 69  |  EiFi.i.i.i.i.i;i 
3C 69 3D 69 3E 69 3F 69  40 69 1E 69 1F 69 20 69  |  <i=i>i?i@i.i.i i 
24 69 25 69 4A 69 4B 69  4C 69 18 69 19 69 1A 69  |  $i%iJiKiLi.i.i.i 
64 69 69 69 6A 69 50 69  51 69 52 69 53 69 56 69  |  diiijiPiQiRiSiVi 
60 69 61 69 08 69 09 69  0A 69 03 69 04 69 05 69  |  `iai.i.i.i.i.i.i 
57 69 7B 69 87 69 88 69  83 69 84 69 8C 69 8D 69  |  Wi{i.i.i.i.i.i.i 
8F 69 90 69 94 69 95 69  99 29 77 29 74 69 9A 69  |  .i.i.i.i.)w)ti.i 
9E 69 9F 69 A3 69 A4 69  A8 69 A9 69 AD 69 AE 69  |  .i.i.i.i.i.i.i.i 
D5 69 D0 69 D1 69 C8 69  CD 69 B2 69 B3 69 B4 69  |  .i.i.i.i.i.i.i.i 
B8 69 C2 69 C5 69 D9 69  DA 69 DD 69 E0 69 E4 69  |  .i.i.i.i.i.i.i.i 
EA 69 ED 69 EE 69 F0 69  F3 69 F5 69 F9 29 8C 69  |  .i.i.i.i.i.i.).i 
FD 69 FE 69 F1 89 05 89  07 89 08 89 0B 89 0E 89  |  .i.i............ 
0F 89 11 89 13 89 14 89  19 89 1C 89 21 89 24 89  |  ............!.$. 
12 89 2D 89 1B 89 2F 89  36 89 3B 89 47 89 48 89  |  ..-.../.6.;.G.H. 
4A 89 4B 89 45 89 4C 89  4D 89 4F 89 54 89 57 89  |  J.K.E.L.M.O.T.W. 
58 89 5F 89 62 89 60 89  64 89 67 89 65 89 69 89  |  X._.b.`.d.g.e.i. 
6C 89 71 89 72 89 76 89  79 89 7A 89 7B 89 7D 89  |  l.q.r.v.y.z.{.}. 
7E 89 80 89 9B 89 9C 89  9E 89 B9 89 BD 89 BE 89  |  ~............... 
BA 89 BB 89 C2 89 C6 89  C7 89 C3 89 C4 89 CF 89  |  ................ 
D6 89 D2 89 D3 89 D5 89  D8 89 DF 89 DB 89 DC 89  |  ................ 
DE 89 E1 89 E5 89 E4 89  E8 89 ED 89 EC 89 EF 89  |  ................ 
F3 89 F2 89 F6 89 FB 89  FA A9 01 A9 C3 A9 C5 A9  |  ................ 
C7 A9 03 A9 04 A9 09 A9  0A A9 0B A9 0C A9 0D A9  |  ................ 
0E A9 11 A9 12 A9 17 A9  18 A9 19 A9 1A A9 1B A9  |  ................ 
1C A9 1D A9 21 A9 10 A9  02 A9 24 A9 A9 A9 AA A9  |  ....!.....$..... 
AB A9 30 A9 31 A9 32 A9  36 A9 37 A9 38 A9 25 A9  |  ..0.1.2.6.7.8.%. 
27 A9 28 A9 2A A9 2B A9  2C A9 3B A9 3C A9 40 A9  |  '.(.*.+.,.;.<.@. 
46 A9 56 A9 57 A9 47 A9  48 A9 4C A9 4D A9 51 A9  |  F.V.W.G.H.L.M.Q. 
52 A9 5A A9 5D A9 5E A9  61 A9 64 A9 65 A9 68 A9  |  R.Z.].^.a.d.e.h. 
6B A9 6E A9 76 A9 80 A9  84 A9 85 A9 89 A9 8B A9  |  k.n.v........... 
94 A9 95 A9 98 A9 9A A9  9E A9 9F A9 A0 A9 A3 A9  |  ................ 
9C A9 AD A9 AE A9 B3 A9  B8 A9 B9 A9 BC A9 BD A9  |  ................ 
BF A9 C0 A9 79 A9 69 49  9A 49 89 49 8B 49 8D 49  |  ....y.iI.I.I.I.I 
94 49 9F 49 83 49 85 49  87 49 81 22 19 12 5B 19  |  .I.I.I.I.I."..[. 
11 ED 19 14 C2 19 14 07  19 15 E8 19 13 63 19 14  |  .............c.. 
72 19 13 8E 19 13 80 19  15 24 19 15 BF 19 12 AE  |  r........$...... 
19 0C A5 19 0C 89 19 0C  FA 19 11 56 19 11 3C 00  |  ...........V..<. 
41 2F 72 6F 6F 74 2F 67  69 74 2F 74 65 73 74 5F  |  A/root/git/test_ 
61 76 6D 2F 5F 62 75 69  6C 64 2F 64 65 66 61 75  |  avm/_build/defau 
6C 74 2F 6C 69 62 2F 67  65 6E 5F 6E 65 74 6C 69  |  lt/lib/gen_netli 
6E 6B 2F 73 72 63 2F 6E  65 74 6C 69 6E 6B 2E 65  |  nk/src/netlink.e 
72 6C 00 4D 2F 72 6F 6F  74 2F 67 69 74 2F 74 65  |  rl.M/root/git/te 
73 74 5F 61 76 6D 2F 5F  62 75 69 6C 64 2F 64 65  |  st_avm/_build/de 
66 61 75 6C 74 2F 6C 69  62 2F 67 65 6E 5F 6E 65  |  fault/lib/gen_ne 
74 6C 69 6E 6B 2F 73 72  63 2F 6E 65 74 6C 69 6E  |  tlink/src/netlin 
6B 5F 64 65 63 6F 64 65  72 5F 67 65 6E 2E 68 72  |  k_decoder_gen.hr 
6C                                                |  l 
```

rtATvw, 2023-12-20:
[test_avm.tar.gz](https://github.com/atomvm/AtomVM/files/13731123/test_avm.tar.gz)

pguyot, 2025-03-16:
I think this was fixed by #1557 . Please reopen if you can still see this error message.

closed 2025-03-16 by pguyot