atomvm/AtomVM #1008: Upload address for atomvm image with ESP32-S3 module

dischoen, 2024-01-03:
Hi,

I wanted to try AtomVM on a ESP32-S3-DEV-KIT-N8R8.
Initially I had errors of the kind

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x13a0
load:0x206f7420,len:0x69726576
Invalid image block, can't boot.
ets_main.c 329
```

For me, the problem was solved by uploading to address 0x0000 instead of 0x1000.
I thought to notify you because the startup guide for ESP32 modules talks about address 0x1000.
(https://www.atomvm.net/doc/master/getting-started-guide.html#flashing-a-binary-image)

Maybe the memory layout in regard to start address is different for various ESP32 modules?

I have an example project here:
https://github.com/dischoen/atomvm_hello_world

fadushin, 2024-01-03:
Excellent, thanks for your input.  We’ll take a look at other modules and maybe create a table in the docs for different ESP modules.

UncleGrumpy, 2024-01-05:
This is definitely something we overlooked in the documentation! I am preparing some documentation updates and will add a table with the bootloader offsets for each chip to the Getting Started guide. Thank you for pointing out this oversight.

commit 63dee030 (2024-02-19): Merge pull request #1045 from UncleGrumpy/docs_updates

Updates to the generated documentation sources

Various updates and corrections top the published documentation. Some
improvements have been made to the rendering of the documentation, including
language highlighting, styled notes, and formatting fixes to make wide text
blocks fit in the display. Links have been added for most function and module
APIs when they are referenced in the guides. Note, these links will not be
enabled until #1037 is merged. They rely on `myst_heading_anchors = 5`

closes #1008

These changes are made under both  [... cut]
PR #1045 (2024-02-19): Updates to the generated documentation sources
Various updates and corrections top the published documentation. Some improvements have been made to the rendering of the documentation, including language highlighting, styled notes, and formatting fixes to make wide text blocks fit in the display. Links have been added for most function and module APIs when they are referenced in the guides. Note, these links will not be enabled until #1037 is merged. They rely on `myst_heading_anchors = 5`

closes #1008

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public Licen [... cut]
closed 2024-02-28 by bettio