atomvm/AtomVM #1016: Potential crash in i2c port driver when using `i2c:write_bytes/2`

fadushin, 2024-01-25:
The current implementation of the `i2c:write_bytes/2' function in the `i2c_driver.c` module uses a direct memory reference from and Erlang binary (which may refer to an address in the process heap, or to a reference counted binary).  However, the memory passed into `i2c_write_bytes` must be valid until the transmission is ended, which could happen after the binary reference is garbage collected.

The solution is to use i2c_write_byte, one byte at a time, from the buffer passed into the driver.

petermm, 2025-03-08:
believe this can be closed by https://github.com/atomvm/AtomVM/pull/1514

pguyot, 2025-03-09:
I'm not sure this bug is closed by this PR but I'm not sure either this bug (still) exists with supported versions of esp-idf SDK.

Fred's concern was that `i2c_master_write` is passed a pointer to a buffer that can be accessed after the function returned, and I understand that he believed it could be called after the binary is GC'd.

My understanding of [esp-idf 4.3 documentation](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/peripherals/i2c.html) is that it's true, but all data from the pointer is read when `i2c_master_cmd_begin` returns. And in our function `i2cdriver_write_bytes` there is no GC between our call to `i2c_master_write` line 522 and our call to `i2c_master_cmd_begin` line 536. Likewise for the resource implementation.

So indeed I think we can close this issue.