atomvm/AtomVM #1077: Problems testing ESP32 and Pico simultaneously

UncleGrumpy, 2024-02-28:
This is not an issue in AtomVM, but it does affect usability and testing.  I have observed that when a pico is plugged into USB first, ESP32 devices plugged in after do not present themselves in /dev and connecting a serial monitor or flashing is not possible. The situation is the same but reversed if an ESP32 device is plugged in first.  This makes monitoring the console impossible when testing network applications between the two platforms.

This is a situation that has been reported by other users, and a workaround, or at least an explanation should be found.

petermm, 2024-02-28:
You also have quirks like an esp32 s3/c3 will be `/dev/cu.usbmodem1101` when connected left side (M1 MB Pro) - but `/dev/cu.usbmodem2101` when connected right side..

UncleGrumpy, 2024-03-04:
I believe I have tracked my problem down to a failing port on a powered USB hub, so I am closing this issue.

closed 2024-03-04 by UncleGrumpy