atomvm/AtomVM #1094: PicoW does not properly set hostname

UncleGrumpy, 2024-03-04:
The pico hostname is not currently being set to `atomvm-{MACADDRESS}` as it should be. The hostname is always being set to `PicoW`. The same behavior is observed if a hostname is supplied in the wifi config. This is consistent when in STA and STA+AP mode.

UncleGrumpy, 2024-03-04:
After some investigating, I realize this feature has not yet been enabled on the picow. I saw the `get_default_device_name` function an assumed it was used later to set the hostname. In fact this is only used as the default ssid in AP mode if no ssid is supplied.

pguyot, 2024-03-04:
I'm not sure what setting the hostname means. Do we run mDNS on the Picow?

UncleGrumpy, 2024-03-04:
Sorry, I am referring to the DHCP client hostname, the name that resolves to the assigned IP address.

UncleGrumpy, 2024-03-04:
When there are multiple pico connected to a wireless network, they all request the same hostname ("PicoW") from the DHCP server, so pinging a pico by name will not work, it will only work if you use the assigned IP.

PR #1563 (2025-06-07): Set pico_w unique dhcp_hostname or configured hostname
Correctly set `dhcp_hostname` from `sta_config_options()` and/or `ap_config_options()`, or, if undefined, sets a unique hostname to the default device name of `atomvm-${DEVICE_MAC}`. Formerly all pico_w devices tried to register to an access point with the factory default `PicoW` hostname, which would make all subsequent pico devices to connect to an access point, after the first one, unreachable by hostname, and cause unpredictable problems for some routers.

Fixes #1094 on release-0.6 branch

IMPORTANT: DO NOT FORWARD this commit to main [... cut]
commit e76ab090 (2025-06-07): Set inique dhcp_hostname, or configured name on pico_w

Correctly set `dhcp_hostname` from `sta_config_options()` and/or `ap_config_options()`, or, if
undefined, sets a unique hostname to the default device name of `atomvm-${DEVICE_MAC}`. Formerly
all pico_w devices tried to register to an access point with the factory default `PicoW` hostname,
which would make all subsequent pico devices to connect to an access point, after the first one,
unreachable by hostname, and cause unpredictable problems for some routers.

Fixes #1094 on release-0.6 branch

IMPORTANT: DO NOT FORWARD this commit to mai [... cut]
closed 2025-06-12 by bettio