atomvm/AtomVM #1112: ESP32-C3 printout of #Ref seems wrong

karlsson, 2024-03-06:
I am running an ESP32-C3 (RISC-V) and when I started logging on debug level I noticed all #Ref types was printed with the same value:  #Ref<0.0.0.16>. Like:
`1970-01-01T00:00:00.854Z [info] <0.10.0> gen_tcp_socket:init/1 (/home/mikael/work/AtomVM/libs/estdlib/src/gen_tcp_socket.erl:215) Starting process in listen (passive) mode.  Socket={<<"">>,#Ref<0.0.0.16>} ControllingProcess=<0.3.0> Options=#{active => true,buffer => 0,inet_backend => socket,timeout => infinity}`

I don't know if this also happens on other ESP32 devices.

When I modify the printout format in nifs.c (and term.c):
```
src/libAtomVM/nifs.c
@@ -3137,7 +3137,7 @@ static term nif_erlang_ref_to_list(Context *ctx, int argc, term argv[])
     #ifdef __clang__
             "#Ref<0.0.0.%llu>";
     #else
-            "#Ref<0.0.0.%lu>";
+            "#Ref<0.0.0.%llu>";
     #endif
     // 2^64 = 18446744073709551616 (20 chars)
     // 12 chars of static text + '\0'
```
It works fine, but I am not sure if this will break other platforms. I guess the ifdef was put there for a reason.
main branch, commit 16748cff, Mar 5

UncleGrumpy, 2024-03-06:
It looks like this might be an issue that came up with version 5.0 that we just haven't caught yet. There were changes to some of the default types. This is documented in the migrating 4.4 to 5.0 guide:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/gcc.html#espressif-toolchain-changes

UncleGrumpy, 2024-03-06:
If you have a 4.4 version of ESP-IDF available could you try building with the unmodified nifs.c and term.c and see if it works as expected with a v4.4 build?

fadushin, 2024-03-06:
The `Socket={<<"">>,#Ref<0.0.0.16>}` is the (opaque) socket -- you are not (generally) supported to peek inside the contents.  The ref part of the opaque type is due to an internal implementation detail, but it is used to uniquely identify different socket instances.

Are you saying that something like:

```
erlang:display(erlang:make_ref()),
erlang:display(erlang:make_ref()),
erlang:display(erlang:make_ref()),
erlang:display(erlang:make_ref()),
erlang:display(erlang:make_ref()),
```

all print the same values?

bettio, 2024-03-06:
There was a bug, as you mentioned the format string was wrong, the correct portable format string is: `"#Ref<0.0.0.%" PRIu64 ">"` which is specific for `uint64_t`.

`%lu` is usually good for unsigned 64-bit integer on 64-bit archs, but that's not the case for ESP32.

#1113 fixes this.

karlsson, 2024-03-07:
I think you have a similar ifdef in term.c function term_funprint.

bettio, 2024-03-07:
> I think you have a similar ifdef in term.c function term_funprint.

yes, you are right, I started working on that, but also I need to remove some hardcoded information. I will try to make that fix by v0.6.1.

karlsson, 2024-03-07:
I reported on main branch but I will close this one since it is now merged to release branch. Thanks!

commit b4c48661 (2024-03-07): Merge pull request #1113 from bettio/fix-term-to-list

Fix `erlang:ref_to_list/1`

Closes #1112

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2024-03-07 by karlsson