atomvm/AtomVM #1118: ESP32-H2 Support

doawoo, 2024-03-15:
Hey there! Long time watcher of this project-- I heard about the amazing CodeBEAM Demos the other week and wanted to check this out again! 

I have a ton of ESP32-H2 boards I wanted to try it on, and I've setup the IDF env properly as far as I can tell- however it seems the final linker step fails complaining about missing symbols like `WIFI_STATUS` ... there's also a few messages that state `IRAM0 segment data does not fit.`

I'll post my full logs below. 
I would also be happy to dig more into possible solutions for this, but just wanted to post this and see if there was existing knowledge of these boards not working with the project :)

UncleGrumpy, 2024-03-15:
We have not implemented support yet for the H2 boards, as you have discovered. We have started working towards supporting them, but they do not support the traditional esp32 WiFi driver. Their marketing point is Thread/Matter support. AtomVM does not currently have any Thread support at all, or Bluetooth for that matter, so even once we can support building for them, there will be a long way to go to fully support these boards. If you want to pursue using them for basic non networked applications the place to start would be conditionally wrapping the WiFi driver so it is not built when the target device is an esp32h2. I haven't tried to do this yet, as I don't have any H2 to test on, but I know other members of the team are poking at it too. Hang in there, we more than likely support them in the near future, and in the meantime the ESP32C6, that we recently added support for, are blazing fast for a single core option.

doawoo, 2024-03-18:
Thanks for the really informative reply! This makes total sense. I'll close this for now since it's known :)

I just ordered a few C6 boards to play with, and in the meantime I have a base ESP32 board that works great.

closed 2024-03-18 by doawoo