atomvm/AtomVM #1161: Possibility of running GenServers in the browser

venkatd, 2024-05-19:
Hi, I saw the lightning talk and that led me here.

I was curious what the feasibility of running Elixir GenServers in the browser is using webassembly.

What is the performance like? How large is the VM payload? Is this compatible with OTP meaning it could interoperate with the BEAM?

pguyot, 2024-05-20:
Hello, thank you for your interest in AtomVM emscripten port.

OTP gen_server can run in the browser using webassembly as demonstrated by the following demo code:
https://github.com/atomvm/atomvm_examples/tree/master/demos/supervised_discs

I guess the same code could be rewritten using Elixir.

Performance is good enough for this example, using several threads (and cores) on the browser. The main slowness in this case is related to DOM manipulation and a better implementation should use HTML5 Canvas for rendering :-)

Interoperation is a different story. Code running in the browser has limited network capabilities (cannot do arbitrary TCP) and AtomVM currently doesn't support disterl. So an Erlang node running in the browser is unlikely. Still, we could use Emscripten fetch API and maybe have the code in the browser interact with an Erlang or Elixir web server. Bindings for FetchAPI are not available but would be very easy to add.

What use case do you have in mind?

venkatd, 2024-05-20:
Thanks!

At this point it is just a curiosity. I thought it would be an interesting use case to be able to write Elixir full stack with normal message passing between processes over the network.