atomvm/AtomVM #1168: ESP: building with v4.4 and v5.0 seem to be failed

takasehideki, 2024-05-26:
For the ESP32 platform, but as the title says, it does not seem to build well (`idf.py build`) with ESP-IDF v4.4 and v5.0. FYI, I am trying it out in an Ubuntu 20.04 based Docker environment.

with ESP-IDF v4.4.4: 
```
[1062/1079] Building C object esp-idf/..._avm_builtins.dir/network_driver.c.obj
FAILED: esp-idf/avm_builtins/CMakeFiles/__idf_avm_builtins.dir/network_driver.c.obj
/root/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DAVM_CREATE_STACKTRACES -DAVM_NO_SMP -DAVM_SELECT_IN_TASK -DAVM_TASK_DRIVER_ENABLED -DHAVE_CLOSE -DHAVE_OPEN -DHAVE_PLATFORM_ATOMIC_H -DHAVE_PTHREAD_RWLOCK -DHAVE_SELECT -DHAVE_SOCKET -DHAVE_SOC_CPU_CORES_NUM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/avm_builtins/include -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/include -I/root/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/root/esp/esp-idf/components/freertos/port/riscv/include -I/root/esp/esp-idf/components/freertos/include/esp_additions -I/root/esp/esp-idf/components/esp_hw_support/include -I/root/esp/esp-idf/components/esp_hw_support/include/soc -I/root/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/root/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -I/root/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/lwip/include/apps -I/root/esp/esp-idf/components/lwip/include/apps/sntp -I/root/esp/esp-idf/components/lwip/lwip/src/include -I/root/esp/esp-idf/components/lwip/port/esp32/include -I/root/esp/esp-idf/components/lwip/port/esp32/include/arch -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/soc/esp32c3/. -I/root/esp/esp-idf/components/soc/esp32c3/include -I/root/esp/esp-idf/components/hal/esp32c3/include -I/root/esp/esp-idf/components/hal/include -I/root/esp/esp-idf/components/hal/platform_port/include -I/root/esp/esp-idf/components/esp_rom/include -I/root/esp/esp-idf/components/esp_rom/include/esp32c3 -I/root/esp/esp-idf/components/esp_rom/esp32c3 -I/root/esp/esp-idf/components/esp_common/include -I/root/esp/esp-idf/components/esp_system/include -I/root/esp/esp-idf/components/esp_system/port/soc -I/root/esp/esp-idf/components/esp_system/port/include/riscv -I/root/esp/esp-idf/components/esp_system/port/public_compat -I/root/esp/esp-idf/components/riscv/include -I/root/esp/esp-idf/components/driver/include -I/root/esp/esp-idf/components/driver/esp32c3/include -I/root/esp/esp-idf/components/esp_pm/include -I/root/esp/esp-idf/components/esp_ringbuf/include -I/root/esp/esp-idf/components/efuse/include -I/root/esp/esp-idf/components/efuse/esp32c3/include -I/root/esp/esp-idf/components/vfs/include -I/root/esp/esp-idf/components/esp_wifi/include -I/root/esp/esp-idf/components/esp_event/include -I/root/esp/esp-idf/components/esp_netif/include -I/root/esp/esp-idf/components/esp_eth/include -I/root/esp/esp-idf/components/tcpip_adapter/include -I/root/esp/esp-idf/components/esp_phy/include -I/root/esp/esp-idf/components/esp_phy/esp32c3/include -I/root/esp/esp-idf/components/esp_ipc/include -I/root/esp/esp-idf/components/app_trace/include -I/root/esp/esp-idf/components/esp_timer/include -I/root/AtomVM/src/libAtomVM -Iesp-idf/libatomvm/libAtomVM -I../components/libatomvm/../avm_sys -I../components/avm_sys/include -I/root/esp/esp-idf/components/spi_flash/include -I/root/esp/esp-idf/components/pthread/include -I/root/esp/esp-idf/components/mbedtls/port/include -I/root/esp/esp-idf/components/mbedtls/mbedtls/include -I/root/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/root/esp/esp-idf/components/nvs_flash/include -march=rv32imc    -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Og -fmacro-prefix-map=/root/AtomVM/src/platforms/esp32=. -fmacro-prefix-map=/root/esp/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.4\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -Werror=incompatible-pointer-types -MD -MT esp-idf/avm_builtins/CMakeFiles/__idf_avm_builtins.dir/network_driver.c.obj -MF esp-idf/avm_builtins/CMakeFiles/__idf_avm_builtins.dir/network_driver.c.obj.d -o esp-idf/avm_builtins/CMakeFiles/__idf_avm_builtins.dir/network_driver.c.obj   -c ../components/avm_builtins/network_driver.c
../components/avm_builtins/network_driver.c: In function 'maybe_set_sntp':
../components/avm_builtins/network_driver.c:536:13: error: implicit declaration of function 'esp_sntp_setoperatingmode'; did you mean 'sntp_setoperatingmode'? [-Werror=implicit-function-declaration]
             esp_sntp_setoperatingmode(SNTP_OPMODE_POLL);
             ^~~~~~~~~~~~~~~~~~~~~~~~~
             sntp_setoperatingmode
../components/avm_builtins/network_driver.c:537:13: error: implicit declaration of function 'esp_sntp_setservername'; did you mean 'sntp_setservername'? [-Werror=implicit-function-declaration]
             esp_sntp_setservername(0, host);
             ^~~~~~~~~~~~~~~~~~~~~~
             sntp_setservername
../components/avm_builtins/network_driver.c:539:13: error: implicit declaration of function 'esp_sntp_init'; did you mean 'esp_wifi_init'? [-Werror=implicit-function-declaration]
             esp_sntp_init();
             ^~~~~~~~~~~~~
             esp_wifi_init
../components/avm_builtins/network_driver.c: In function 'get_sta_rssi':
../components/avm_builtins/network_driver.c:771:21: error: implicit declaration of function 'esp_wifi_sta_get_rssi'; did you mean 'esp_wifi_set_csi'? [-Werror=implicit-function-declaration]
     esp_err_t err = esp_wifi_sta_get_rssi(&sta_rssi);
                     ^~~~~~~~~~~~~~~~~~~~~
                     esp_wifi_set_csi
cc1: some warnings being treated as errors
[1067/1079] Building C object esp-idf/...f_avm_builtins.dir/socket_driver.c.obj
../components/avm_builtins/socket_driver.c: In function 'do_receive_data':
../components/avm_builtins/socket_driver.c:658:39: warning: unused variable 'platform' [-Wunused-variable]
             struct ESP32PlatformData *platform = ctx->global->platform_data;
                                       ^~~~~~~~
../components/avm_builtins/socket_driver.c: In function 'do_close':
../components/avm_builtins/socket_driver.c:1245:31: warning: unused variable 'platform' [-Wunused-variable]
     struct ESP32PlatformData *platform = ctx->global->platform_data;
                               ^~~~~~~~
[1070/1079] Building C object esp-idf/...CMakeFiles/libAtomVM.dir/context.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

with ESP-IDF v5.0: 
```
[94/94] cd /root/AtomVM/src/platforms/.../esp32/build/bootloader/bootloader.binBootloader binary size 0x4ed0 bytes. 0x3130 bytes (38%) free.
[835/864] Building C object esp-idf/av.../__idf_avm_sys.dir/platform_nifs.c.objFAILED: esp-idf/avm_sys/CMakeFiles/__idf_avm_sys.dir/platform_nifs.c.obj
/root/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DAVM_CREATE_STACKTRACES -DAVM_NO_SMP -DAVM_SELECT_IN_TASK -DAVM_TASK_DRIVER_ENABLED -DHAVE_CLOSE -DHAVE_OPEN -DHAVE_PLATFORM_ATOMIC_H -DHAVE_SELECT -DHAVE_SOCKET -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/avm_sys/include -I/root/esp/esp-idf/components/newlib/platform_include -I/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/root/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/root/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/root/esp/esp-idf/components/freertos/esp_additions/include -I/root/esp/esp-idf/components/esp_hw_support/include -I/root/esp/esp-idf/components/esp_hw_support/include/soc -I/root/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/root/esp/esp-idf/components/esp_hw_support/port/esp32c3/. -I/root/esp/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/root/esp/esp-idf/components/heap/include -I/root/esp/esp-idf/components/log/include -I/root/esp/esp-idf/components/soc/include -I/root/esp/esp-idf/components/soc/esp32c3/. -I/root/esp/esp-idf/components/soc/esp32c3/include -I/root/esp/esp-idf/componen [... cut ...]
Rebasing these changes may eliminate this problem, or there may be other better solutions. However, I think that these coping strategies may hinder the growth of technology.

My recommended conclusion is that users who want to build AtomVM virtual machine by themselves should use ESP-IDF v5.1 or later, with proper documentation. I think the following document needs to be modified.
https://www.atomvm.net/doc/main/release-notes.html#esp32-support
https://github.com/atomvm/AtomVM/blob/main/doc/release-notes.md.in#L63-L70

What do you think about my proposal?
I'll be happy if you want me to create PR for the modification of docs..

petermm, 2024-05-26:
apologies for you not having the best experience.

CI is running 4.4.7, 5.0.6, 5.1.3, 5.2.1, esp32c3 5.2.1: https://github.com/atomvm/AtomVM/actions/runs/9238140806/job/25415891557

So seems like we have hit a snag against older 4.x 5.x SDKs.

If it's not too time consuming can you try using 4.4.7 and 5.0.6?

I will investigate further..

takasehideki, 2024-05-27:
@petermm thank you so much for your reply.
You have nothing to apologize for. I'd be happy to contribute to a better development experience.

So I wasn't aware of the CI build. I have tried with 4.4.7 and 5.0.6 and was able to build successfully!
Also FYI, I forgot to tell you an important piece of information: I was doing it for `esp32c3` (but I think that ESP-IDF does not have much influence on the differences between the different targets).

According to my experience, I think it is better to include the subminor number in the document. What do you think?

takasehideki, 2024-06-05:
#1170 has been merged, so I close this!!

commit 16c7096c (2024-06-04): Merge pull request #1170 from takasehideki/improve_release_note

Improve release notes for ESP32 support

This PR tries to include the subminor number of ESP-IDF supported versions
because it is sensitive to build firmware for the ESP32 family. Also, v5.2
(v5.2.1) is covered at this time, so this has been added.
Moreover, I added link to target `idf-version` on
[esp32-build.yaml](https://github.com/atomvm/AtomVM/blob/main/.github/workflows/esp32-build.yaml#L38).
As future development progresses, the information in this table may diverge
from the CI build target numbers, but I expect that users [... cut]
PR #1170 (2024-06-04): Improve release notes for ESP32 support
This PR tries to include the subminor number of ESP-IDF supported versions because it is sensitive to build firmware for the ESP32 family. Also, v5.2 (v5.2.1) is covered at this time, so this has been added.
Moreover, I added link to target `idf-version` on [esp32-build.yaml](https://github.com/atomvm/AtomVM/blob/main/.github/workflows/esp32-build.yaml#L38). As future development progresses, the information in this table may diverge from the CI build target numbers, but I expect that users can follow this link to the latest status.

This will close #1168 [... cut]
closed 2024-06-05 by takasehideki