atomvm/AtomVM #1169: suggestion: delete platform dependent `README.*.Md`

takasehideki, 2024-05-26:
This repository has the methods used for each platform, such as `README.ESP32.Md`.
In relation to #1168, I initially followed this procedure, not realizing that there is now an [official documentation page](https://www.atomvm.net/doc/main/index.html) (maintained under `doc/`), and it took me a while to realize that my procedure was not right. 
Also, there seems to be an inconsistency between the procedures described in these `README.*.Md` and the official page. I understand that it is very difficult to keep up with the multiple documents that exist,,,;(

Therefore, I want to suggest that these `README.*.Md` should be deleted. I think it will be necessary to add a notice to the deletion.

About this proposal along with #1168, I'll also be happy if you want me to create PR for the modification of docs.

UncleGrumpy, 2024-05-31:
I completely agree! These platform specific README files were created as a basic roadmap for getting started with each platform, but most are missing important steps, and leave a lot of assumptions to the user. Since their creation we have added detailed setup and usage procedures for each platform in the official documentation. It does not make sense to maintain two copies of this information that need to be kept in sync, and these README files have caused quite a bit of confusion for users. I do believe removing them is the best course of action. The main project README already includes links to the appropriate getting started section for each platform and users should look there for guidance to flash the VM binary and user applications.

takasehideki, 2024-06-05:
Thank you so much for the positive response!
I created #1179 to fix this situation, so please review and comment there.

commit b4a413c4 (2024-06-10): Merge pull request #1179 from takasehideki/release-0.6_del_platform_dependent_readme

Delete platform dependent README.*.Md

fix #1169

This PR will delete `README.*.Md`, which explained how to use AtomVM on each
platform.
Since these are now described on the
[official documentation page](https://www.atomvm.net/doc/main/index.html), I
recommend removing them to avoid confusion for first-time visitors.
In addition, I modified the link to each README to the directories where the
implementations for each platform are located (I think this will be useful for
users).
Also, links to the Build Instru [... cut]
PR #1179 (2024-06-10): Delete platform dependent README.*.Md
fix #1169 

This PR will delete `README.*.Md`, which explained how to use AtomVM on each platform.
Since these are now described on the [official documentation page](https://www.atomvm.net/doc/main/index.html), I recommend removing them to avoid confusion for first-time visitors.
In addition, I modified the link to each README to the directories where the implementations for each platform are located (I think this will be useful for users).
Also, links to the Build Instructions page were duplicated, so I organized them.

---
These changes are made under bo [... cut]
closed 2024-06-27 by bettio