atomvm/AtomVM #1174: Trying to port elixir library

Ridtt, 2024-05-30:
Hello, im trying to compile elixir library into .avm pack using mix atomvm.packbeam

Only errors left I have is - 

error: following modules or functions are not available on AtomVM:
* Elixir.Enum:into/2
* Elixir.RuntimeError:exception/1

If Enum.into could be replaced possible (or better implemented?), but how to raise erl exception instead of elixir exception?

UncleGrumpy, 2024-05-30:
You should be able to use an erlang:error/1 or erlang:throw/1.

UncleGrumpy, 2024-05-30:
> I have big hope to run my existing binary structures decoding/encoding library in targets like wasm and stm32.

Not to diminish your hopes, but both of those platforms have limited support. Depending on what you are hoping to accomplish they may well work for your needs, but the esp32 platform has the best level of support, and the pico(w) is also quite capable for networking and basic gpio. STM32 has gpio support but no networking yet, and no I2C or SPI, but if you are familiar with the platform any contributions are more than welcome.

UncleGrumpy, 2024-05-30:
> You should be able to use an erlang:error/1 or erlang:throw/1.

Actually from looking at the docs Erlang:raise/1 is probably what you are looking for.

https://hexdocs.pm/elixir/main/RuntimeError.html

Ridtt, 2024-07-01:
@UncleGrumpy 
Thank you for such a warm answer! I want you to know I decided not dig into your great project more atm, there is lack of what I need currently. I will have my eyesight on it, project idea is really brave  thing to even imagine, you have my respect.

In real word scenario from my perspective besides implementation of all vm opcodes would be super interesting to have an option to write generic implementation or just declare i need to include it from elixir codebase ether in elixir (I need Enum.into to compile deps, for example) or provide native language implementations, mix of something like embassy/rtic and rustler. Thank you again for what you are doing and dreaming, best regards. 

A

Ridtt, 2024-07-01:
@UncleGrumpy 

I just don’t really understand needs for i2c be part of this project, rtic/embassy have all required drivers for example, such wrapper could be easily just imported by end user. But what could be not is microseconds interruption handling, circular DAC multiplexing on multichannel and every non trivial work every coprocessor for example want to do. What is the reason repeat all those spi/i2c’s etc if it’s not a game changer? I see only way to change this into flawless victory is give user easy and known way to integrate native parts or whole underlying native framework.

commit abf59d86 (2024-08-15): Merge pull request #1237 from bettio/increase-elixir-support

Increase Elixir support

Increases Elixir support by adding support to Enumerable and Collectable
protocols, and to a number of Exceptions.

Fixes #1174

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #1237 (2024-08-15): Increase Elixir support
Increases Elixir support by adding support to Enumerable and Collectable protocols, and to a number of Exceptions.

Fixes #1174

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later