atomvm/AtomVM #1219: test-erlang -b test_gc fails for Erlang/OTP 27.0

karlsson, 2024-06-28:
Erlang/OTP version 27.0

When executing the test_gc case on the BEAM (with -b switch), the testcase fails upon comparing heap_sizes as they are the same before and after garbage collection.
Changing the selector from `heap_size` to `total_heap_size` in the test case makes it pass for both BEAM and AtomVM executions.

pguyot, 2024-06-28:
True. We've been disabling this test in the CI (as well as prime_smp which doesn't work well with GitHub) since:

https://github.com/atomvm/AtomVM/commit/7ef4534a6
https://github.com/atomvm/AtomVM/pull/1013

And I think I blindly copied the line from OTP26 when adding OTP27.
It looks like a quick fix, I don't remember discussing this with Fred. We probably should fix this as you suggest.

karlsson, 2024-06-28:
Well, it actually points to some changes in Erlang/OTP 27 and I haven't found any documented changes for heap, heap_size or garbage collection in the upgrade notes for OTP. So adapting the testcase to get it to work may be leading wrong (although AtomVM is not involved at all :smile: ).
I could check, and possibly rise a question on, the Erlang forum?

karlsson, 2024-06-28:
Added [question](https://erlangforums.com/t/process-info-pid-heap-size-changed-between-otp-26-and-27/3791) on Erlang forum.

karlsson, 2024-06-28:
I tested a bit more and it turns out that if I remove the last dot "." from the eval part in `test.c` it works. 
```
diff --git a/tests/test.c b/tests/test.c
index 48d0399d..3180d52a 100644
--- a/tests/test.c
+++ b/tests/test.c
@@ -587,7 +587,7 @@ static int test_beam(struct Test *test)
         " R =:= %i -> 0;"
         " true -> io:format(\"Expected ~B, got ~p\n\", [%i, R]) "
         "end, "
-        "erlang:halt(S).' -noshell",
+        "erlang:halt(S)' -noshell",
         test->test_module,
         test->expected_value,
         test->expected_value);
```
I do not really know why it works :thinking: but the examples in the erlang docs are without ending ".".
Maybe one should catch the result since you really intend to return 0 for success or an integer > 0 otherwise from `erlang:halt(S)`?

```
static int test_beam(struct Test *test)
{
    char command[512];
    snprintf(command, sizeof(command),
        "erl -pa . -noshell -eval '"
        "erlang:process_flag(trap_exit, false), \n" /* init(3) traps exists */
        "S = case (catch %s:start()) of "
        "  %i -> 0;"
        "  {\'EXIT\',{Error,[MFA|_]}} -> "
        "  io:format(\"Expected ~B, got EXIT ~p~n\", [%i, {Error,[MFA]}]), 1;"
        "  R -> io:format(\"Expected ~B, got ~p~n\", [%i, R]), 1 "
        "end, \n"
        "erlang:halt(S)'",
        test->test_module,
        test->expected_value,
        test->expected_value,
        test->expected_value);
    return system(command);
}
```

commit 10a8f42a (2024-06-29): Merge pull request #1221 from pguyot/w26/fix-test_gc-beam

Fix test_gc test with BEAM

`test_gc` was disabled because it failed more or less randomly with OTP26. It
seems to be related to a change in how the eval string from the command line is
processed by OTP, allocating more memory and therefore making the effect of
garbage collector less obvious. Fix it by spawning a fresh process for the test.

Special thanks to @karlsson for investigating this.

Fixes #1219

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual lic [... cut]
PR #1221 (2024-06-29): Fix test_gc test with BEAM
`test_gc` was disabled because it failed more or less randomly with OTP26. It seems to be related to a change in how the eval string from the command line is processed by OTP, allocating more memory and therefore making the effect of garbage collector less obvious. Fix it by spawning a fresh process for the test.

Special thanks to @karlsson for investigating this.

Fixes #1219

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-l [... cut]
closed 2024-06-29 by bettio