atomvm/AtomVM #1257: `gpiodriver_set_int` does not allow to set interrupt for pin 2

rafaltrojniak, 2024-09-02:
Hello,

I was trying to play around with [Interrupt Handling](https://www.atomvm.net/doc/master/programmers-guide.html#interrupt-handling) on my ESP32. I did run the example form documentation, and I'm getting `error` as an return value from `gpio:set_int/3` every time.

After long debugging, I found out that the  [`gpiodriver_set_int`](https://github.com/atomvm/AtomVM/blob/main/src/platforms/esp32/components/avm_builtins/gpio_driver.c#L427)  fails at step when it checks for [`term_is_integer`](https://github.com/atomvm/AtomVM/blob/main/src/libAtomVM/term.h#L438).

`term_is_integer` receives my pin `2` as just `0x02` binary, and fails the test for being integer.

I'm no expert on the way terms are managed in the atomvm and if the `term_is_integer` is correct.
Is the term here passed in the wrong format for some reason, or is the function not doing the check properly ?

My code is:
```erlang
start() ->
    Pin = 2,
    GPIO = gpio:open(),
    gpio:set_direction(GPIO, Pin, input),
    io:format("Setting interruption for : ~p~n", [Pin]),
    Result = gpio:set_int(GPIO, Pin, rising),
    io:format("When setting pin ~p interrupt got: ~p~n", [Pin, Result]),
```

And getting that as an result:
```
Setting interruption for : 2
When setting pin 2 interrupt got: error
```

UncleGrumpy, 2024-09-02:
If this is an esp32 classic, then pin 2 is the en/reset pin. This is not a valid gpio input pin, it is used to disable the cpu if it is physically pulled low.

UncleGrumpy, 2024-09-02:
One of the open issues is to make the return errors for the esp32 match other platforms by returning {error, Reason}, which would make this marginally more helpful because it would return {error, badarg}, perhaps we should also add better console logs when an error is returned.

UncleGrumpy, 2024-09-02:
I will look into this. At a glance the code around checking the pin in gpiodriver_set_int looks fine, but the way the trigger term is handled looks highly suspicious to me.

rafaltrojniak, 2024-09-03:
https://github.com/atomvm/AtomVM/pull/1258 fixes this issue

UncleGrumpy, 2024-10-12:
It should be safe to close this now. #1258 was merged, it just did not include a closes tag to close it automatically.

closed 2024-10-12 by UncleGrumpy