atomvm/AtomVM #1260: Checksum against truncated / corrupted avm files

bettio, 2024-09-08:
There is no easy way to know if a .avm file is truncated or overwritten with other data.
When this kind of issues happen they might be really annoying to debug. 
So .avm structure and content should be checked for integrity.

There are 2 possible options:
- iterate all .avm sections checking for consistency (and maybe for an end-of-avm section)
- add a checksum at the end of the AVM pack and verify it

UncleGrumpy, 2024-09-08:
I love the idea of adding a checksum for verification. This will make verifying updates over OTA much more reliable.

pguyot, 2024-09-10:
Something that popped out at work (not that we're using AtomVM, though) is the ability to cryptographically verify firmware signatures before applying OTA updates.