atomvm/AtomVM #1262: esp32boot.avm got bigger than the partition that contains it

bettio, 2024-09-08:
The recent addition of a number of Elixir modules likely made esp32boot.avm bigger than the partition that contains it.
It must be shrinked or partitioning schema must be reviewed.

UncleGrumpy, 2024-09-09:
It won't solve the problem but we can safely remove `libs/eavmlib/src/network_fsm.erl`, this module has been documented as deprecated since before the v0.5.0 release.

In the interest of supporting OTA updates this might be a good time to add a second application partition and adjust the size of the `boot.avm` partition. I would propose the following layout:
```
# Name,   Type, SubType,   Offset,      Size,  Flags
nvs,      data, nvs,       0x9000,     0x6000,
phy_init, data, phy,       0xf000,     0x1000,
factory,  app,  factory,  0x10000,   0x1C0000,
boot.avm, data, phy,     0x1D0000,    0x60000,
app1.avm, data, phy,     0x230000,    0xE8000,
app2.avm, data, phy,     0x318000,    0xE8000
```
This increases the `boot.avm` partition from 256k to 512k, and gives each application partition approximately 928k which is only slightly smaller than the size currently used by our single `main.avm` application partition.

Another option would be to make the `boot.avm` even smaller (just including `esp32boot` and `esp32devmode`), and publish `atomvmlib.avm` to hex and have users include it as a dependency, much like the stm32 platform does now. The `atomvm_packbeam` tool does a fine job of pruning unnecessary modules if the `--prune` option is used. If we take this approach the default behavior of the `atomvm_rebabr3_plugin` can be changed to have the packbeam task prune unless an option is passed to prevent it.

bettio, 2024-09-11:
I tried to figure out every bit we can save, and reality is that we cannot fit anyway: we get very close, but we cannot fit into boot partition anyway. Even if we can make it, as soon as we merge any fix or small addition (such as adc PR) we run out of space again.

Given this I rather push for a breaking change limited to Elixir, we'll properly document this, and we'll make sure esp32-mkimage.yaml will make both erlang only and elixir enabled images with 2 different partitioning schemas.

We'll make some kind of different discussion about what we should do for main.

UncleGrumpy, 2024-09-11:
This sounds like a perfectly reasonable short term solution to me.

commit 2aaf7a3c (2024-09-14): Merge pull request #1266 from bettio/esp32-with-bigger-elixir-partition

ESP32: move Elixir support to new Elixir images

**BREAKING CHANGE**:
Starting with this PR, Elixir support is only available in Elixir images with
bigger boot.avm partition.

Fixes #1262

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #1266 (2024-09-14): ESP32: move Elixir support to new Elixir images
**BREAKING CHANGE**:
Starting with this PR, Elixir support is only available in Elixir images with bigger boot.avm partition.

Fixes #1262 

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later