atomvm/AtomVM #1303: Cannot make a packbeam file properly using PackBEAM

aiotter, 2024-10-01:
I usually use `exatomvm` to build a packbeam file. This time I tried to build it with PackBEAM command in order to run the packbeam file on AtomVM newer than v0.6.4, but I wasn't able to.

Firstly, I built AtomVM v0.6.4 inside Docker container.
```dockerfile
FROM elixir:1.17.2

WORKDIR /project/build
RUN apt-get -qq update && apt-get install -qq cmake gperf
RUN curl -sL https://github.com/atomvm/AtomVM/archive/refs/tags/v0.6.4.tar.gz | tar -xz -C /project --strip-components=1
RUN cmake .. && make -j 8 && make install
RUN install /project/build/tools/packbeam/PackBEAM /usr/local/bin/packbeam
```

Compiled a simple module, packed it with the packbeam utility, and then tried to run it. Failed.
```bash
root@7bfb20ff3e74:/workdir# erlc entrypoint.erl
root@7bfb20ff3e74:/workdir# packbeam -a out.avm entrypoint.beam
root@7bfb20ff3e74:/workdir# packbeam -l out.avm
entrypoint.beam  # <- no star * mark
root@7bfb20ff3e74:/workdir# atomvm out.avm
Unable to locate entrypoint.
```

`entrypoint.erl`:
```erl
-module(entrypoint).
-export([start/0]).

start() ->
    erlang:display("Hello, world!").
```

UncleGrumpy, 2024-10-01:
The PackBEAM tool in the AtomVM repo is still used to pack our libraries, but for most user need we offer a more full featured version of packbeam written in Erlang, it is also used by the atomvm_rebar3_plugin. It is the [atomvm—packbeam](https://github.com/atomvm/atomvm_packbeam), which offers a few extra features, like including external AVMs, listing the contents of a packed AVM file. This tool also includes line number details for Erlang stacktraces by default (they can also be omitted). This is the tool we recommend for most users AVM creation and manipulation needs. You may want to reed the section on the tooling we offer for developers working with AtomVM:

https://www.atomvm.net/doc/main/atomvm-tooling.html

aiotter, 2024-10-01:
@UncleGrumpy Thank you for your support. Actually that is what I tried first.

Using that tool I was able to get a runnable avm file, however, some functions in the standard library are unavailable.
I thought there is a chance that atomvm-packbeam is somehow linked to the current release version of the AtomVM, as `exatomvm` does. That is why I tried to make use of PackBEAM.

Can you check that I created the avm file in a correct way?

```bash
root@3ec293897e75:/workdir# packbeam create -s Elixir.TestEntrypoint tests.avm Elixir.TestEntrypoint.beam Elixir.RemoteIdTest.beam Elixir.Hexadecimal.beam Elixir.Parser.RemoteId.beam /project/build/libs/etest/src/etest.avm /project/build/libs/atomvmlib.avm
root@35be96c840bf:/workdir# atomvm tests.avm
Warning: function erlang:binary_to_integer/2 cannot be resolved.
-
[{Elixir.RemoteIdTest,{exception,{undef,[{Elixir.Hexadecimal,-load/1-lbc$^0/2-0-,2,[{file,"lib/formatter.ex"},{line,28}]},{Elixir.RemoteIdTest,test,0,[{file,"atomvm-test/parser.ex"},{line,8}]},{etest,do_run_test,1,[{file,"/project/libs/etest/src/etest.erl"},{line,183}]},{etest,do_run_test,1,[{file,"/project/libs/etest/src/etest.erl"},{line,196}]},{etest,-run_test/1-fun-0-,2,[{file,"/project/libs/etest/src/etest.erl"},{line,165}]}]}}}]
Return value: {fail,{Elixir.RemoteIdTest,{exception,{undef,[{Elixir.Hexadecimal,-load/1-lbc$^0/2-0-,2,[{file,"lib/formatter.ex"},{line,28}]},{Elixir.RemoteIdTest,test,0,[{file,"atomvm-test/parser.ex"},{line,8}]},{etest,do_run_test,1,[{file,"/project/libs/etest/src/etest.erl"},{line,183}]},{etest,do_run_test,1,[{file,"/project/libs/etest/src/etest.erl"},{line,196}]},{etest,-run_test/1-fun-0-,2,[{file,"/project/libs/etest/src/etest.erl"},{line,165}]}]}}}}
```

I believe `atomvmlib.avm` is the file where standard library is packed into, but in case I'm wrong with it I also tried `find /project/build/libs . -name '*.avm' -or -name '*.beam' -exec packbeam create -s Elixir.TestEntrypoint -a tests.avm {} +`. However, the same problem occurred.

<details>
<summary>Dockerfile to install atomvm-packbeam and AtomVM</summary>

```dockerfile
FROM elixir:1.17.2

# Install rebar3
WORKDIR /rebar3
RUN curl -sL https://github.com/erlang/rebar3/archive/refs/tags/3.24.0.tar.gz | tar -xz --strip-components=1
RUN ./bootstrap
RUN cp /rebar3/rebar3 /usr/local/bin

# Install atomvm_packbeam
WORKDIR /atomvm_packbeam
RUN curl -sL https://github.com/atomvm/atomvm_packbeam/archive/refs/tags/0.7.3.tar.gz | tar -xz --strip-components=1
RUN curl -sL https://github.com/atomvm/atomvm_packbeam/commit/65e9053a99912c9550e639d1e7f299d01ef6448c.diff | patch -p1
RUN rebar3 as prod release && rebar3 as prod tar && ./install.sh /usr/local 0.7.3

# Build AtomVM
WORKDIR /project/build
RUN apt-get -qq update && apt-get install -qq cmake gperf
RUN curl -sL https://github.com/atomvm/AtomVM/archive/refs/heads/release-0.6.tar.gz | tar -xz -C /project --strip-components=1
RUN cmake .. && make -j 8 && make install
```
</details>

UncleGrumpy, 2024-10-01:
We do not support the complete OTP. There is just no way to fit this onto the flash of a microcontroller.

If there are standard library functions that you need to use that are not yet supported by AtomVM, I suggest opening an issue for the missing function. Our approach so far has been to implement most of the commonly used functions, and add more as the need arises.

We are happy to accept PRs adding standard functions. We are also willing to make these additions for you if you open an issue explaining your needs, but we are a small team of volunteers, so we cannot make any guarantees about how fast we can implement your feature request ;-)

UncleGrumpy, 2024-10-01:
If this is an esp32 device you might have better luck packing esp32boot.avm instead of atomvmlib.avm… atomvmlib use to be all you needed, but the esp32 platform works a bit differently, so to get the complete standard AotomVM libraries you need esp32boot.avm on that platform. I hope this helps you. 🤞

aiotter, 2024-10-02:
> We do not support the complete OTP. There is just no way to fit this onto the flash of a microcontroller.

Of course you don't, but `erlang:binary_to_integer/2` is already implemented on branch `release-0.6`, isn't it?
https://github.com/atomvm/AtomVM/blob/7d63dc0da6e50413b179121c32c79023570e1fc6/src/libAtomVM/nifs.c#L1817-L1844

I didn't try `esp32boot.avm` yet. I'll try it. But I want to test the code on Unix machine as long as the code does not include ESP32-specific features so we can test it on CI.

UncleGrumpy, 2024-10-02:
Yes, of course, for Linux you should not use esp32boot.avm, atomvmlib.avm is sufficient for the generic_unix port.

Yes, erlang:binary_to_integer/2 is supported. Are you using release binaries, or are you building from source? I am also confused why binary_to_integer/2 isn't being found.

aiotter, 2024-10-02:
I'm building it from source. I made a reproductive Dockerfile. Hope it helps.

```dockerfile
FROM elixir:1.17.2

# Install rebar3
WORKDIR /rebar3
RUN curl -sL https://github.com/erlang/rebar3/archive/refs/tags/3.24.0.tar.gz | tar -xz --strip-components=1
RUN ./bootstrap
RUN cp /rebar3/rebar3 /usr/local/bin

# Install atomvm_packbeam
WORKDIR /atomvm_packbeam
RUN curl -sL https://github.com/atomvm/atomvm_packbeam/archive/refs/tags/0.7.3.tar.gz | tar -xz --strip-components=1
RUN curl -sL https://github.com/atomvm/atomvm_packbeam/commit/65e9053a99912c9550e639d1e7f299d01ef6448c.diff | patch -p1
RUN rebar3 as prod release && rebar3 as prod tar && ./install.sh /usr/local 0.7.3

# Build AtomVM from release-0.6 branch
WORKDIR /atomvm/build
RUN apt-get -qq update && apt-get install -qq cmake gperf
RUN curl -sL https://github.com/atomvm/AtomVM/archive/refs/heads/release-0.6.tar.gz | tar -xz -C /atomvm --strip-components=1
RUN cmake .. && make -j 8 && make install

# Create Elixir module
WORKDIR /workdir
RUN (cat >entrypoint.ex && elixirc entrypoint.ex) <<EOF
defmodule Entrypoint do
  def start(), do: :erlang.binary_to_integer("16", 16) |> :erlang.display()
end
EOF

RUN packbeam create test.avm Elixir.Entrypoint.beam /atomvm/build/libs/atomvmlib.avm

CMD ["atomvm", "test.avm"]
```

btw I found a problem on atomvm-packbeam v0.7.3 and reported it on https://github.com/atomvm/atomvm_packbeam/issues/19 just now.

UncleGrumpy, 2024-10-02:
Sorry, I went back and amended my previous post about binary_to_integer... while you were already answering it ;-)

aiotter, 2024-10-02:
Oh no, arity is different. What a silly mistake I did!
Thank you so much. You saved me.

Okay I'll implement `binary_to_integer/2` then. This is needed to implement `binary:decode_hex/1` (#1287).

closed 2024-10-02 by aiotter