atomvm/AtomVM #1314: Test v0.6.5

bettio, 2024-10-12:
Let's run a number of manual tests and additional checks in order to make sure v0.6.5 will be the best release ever.

Test binaries can be found here: https://github.com/bettio/avm_tag_ci/releases/tag/v0.6.202410120

**Please, let's use real hardware.**

## General

- [ ] Can be compiled with C++ libraries

## ESP32

Devices:

- [ ] boots on ESP32
- [ ] boots on ESP32-S2
- [ ] boots on ESP32-S3
- [ ] boots on ESP32-C2
- [ ] boots on ESP32-C3
- [ ] boots on ESP32-C6
- [ ] boots on ESP32-H2

Network features:
- [ ] Connects to wifi AP
- [ ] Works as wifi AP
- [ ] ssl works
- [ ] gen_tcp works
- [ ] gen_udp works
- [ ] socket works 

Peripherals:

- [ ] ADC read
- [ ] ledc
- [ ] GPIO read
- [ ] GPIO write
- [ ] i2c read
- [ ] i2c write
- [ ] SPI read
- [ ] SPI write
- [ ] UART read
- [ ] UART write

Misc:

- [ ] SD with SPI can be mounted
- [ ] SD with MMC can be mounted
- [ ] SPI flash partition can be mounted  

## RP2040

- [ ] Boots
- [ ] Connects to wifi AP

## emscripten

- [ ] Supervisor bubbles tests works

## generic_unix

## STM32

bettio, 2024-10-12:
Tested i2c, gpio interrupt and atomgl (with ili934x driver) using an Elixir application.
Everything work as expected.

Log:
```
load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (31) boot: compile time Oct 12 2024 17:46:24
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 001c0000
I (92) boot:  3 boot.avm         RF data          01 01 001d0000 00080000
I (99) boot:  4 main.avm         RF data          01 01 00250000 00100000
I (107) boot: End of partition table
I (111) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=39748h (235336) map
I (200) esp_image: segment 1: paddr=00049770 vaddr=3ff80063 size=00008h (     8) load
I (200) esp_image: segment 2: paddr=00049780 vaddr=3ffb0000 size=04390h ( 17296) load
I (212) esp_image: segment 3: paddr=0004db18 vaddr=40080000 size=02500h (  9472) load
I (218) esp_image: segment 4: paddr=00050020 vaddr=400d0020 size=1351d0h (1266128) map
I (657) esp_image: segment 5: paddr=001851f8 vaddr=40082500 size=19ba8h (105384) load
I (698) esp_image: segment 6: paddr=0019eda8 vaddr=400c0000 size=00064h (   100) load
I (713) boot: Loaded app from partition at offset 0x10000
I (713) boot: Disabling RNG early entropy source...
I (725) cpu_start: Multicore app
I (733) cpu_start: Pro cpu start user code
I (733) cpu_start: cpu freq: 160000000 Hz
I (734) app_init: Application information:
I (736) app_init: Project name:     atomvm-esp32
I (742) app_init: App version:      1
I (746) app_init: Compile time:     Oct 12 2024 17:46:18
I (752) app_init: ELF file SHA256:  881f099d8...
I (757) app_init: ESP-IDF:          v5.3.1
I (762) efuse_init: Min chip rev:     v0.0
I (767) efuse_init: Max chip rev:     v3.99 
I (772) efuse_init: Chip rev:         v3.0
I (777) heap_init: Initializing. RAM available for dynamic allocation:
I (784) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (790) heap_init: At 3FFB9528 len 00026AD8 (154 KiB): DRAM
I (796) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (802) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (809) heap_init: At 4009C0A8 len 00003F58 (15 KiB): IRAM
I (817) spi_flash: detected chip: generic
I (820) spi_flash: flash io: dio
W (824) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (835) main_task: Started on CPU0
I (845) main_task: Calling app_main()

    ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ## 
      ## ##      ##    ##     ## ###   ### ##     ## ###   ### 
     ##   ##     ##    ##     ## #### #### ##     ## #### #### 
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ## 
    #########    ##    ##     ## ##     ##  ##   ##  ##     ## 
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ## 
    ##     ##    ##     #######  ##     ##    ###    ##     ## 

    ###########################################################

I (885) AtomVM: Starting AtomVM revision 0.6.5
I (895) sys: Loaded BEAM partition boot.avm at address 0x1d0000 (size=524288 bytes)
I (925) network_driver: Initialized network interface
I (925) network_driver: Created default event loop
I (935) AtomVM: Found startup beam esp32init.beam
I (945) AtomVM: Starting esp32init.beam...
---
AtomVM init.
I (955) sys: Loaded BEAM partition main.avm at address 0x250000 (size=1048576 bytes)
Starting application...
```

UncleGrumpy, 2024-10-12:
I discovered a problem with `gen_udp`, not sure if it is a problem on the sending or receiving side, but the server never acknowledges receiving a packet.

bettio, 2024-10-12:
Opened #1318 after @UncleGrumpy ESP32-H2 tests.

pguyot, 2024-10-13:
I don't seem to be able to boot esp32s2 regular (non-elixir) image. Elixir image looks ok (Wifi shows up and I can connect to HTTP and alisp servers).

For the record, ESP32S2 (at least the board I have) doesn't feature serial over USB unless explicitely enabled in software with tinyusb, so it's difficult to debug and our builds don't have this by default (and even if they did, the issue seems to occur before it is enabled).

I suspect a partition overflow issue.

pguyot, 2024-10-13:
Pico-W works as wifi AP

pguyot, 2024-10-13:
I'm not able yet to connect Pico-W on my phone's personal hotspot. It doesn't say it got an IP adress.

pguyot, 2024-10-13:
Pico-W does connect to AtomVM-ESP32 AP from ESP32S2.

pguyot, 2024-10-13:
esp32devmode doesn't seem to allow configuration of SSID/PSK. This is regular image on an ESP32S3.

```
Welcome to minicom 2.9

OPTIONS: I18n 
Compiled on Oct  4 2023, 07:48:07.
Port /dev/cu.usbmodem14301, 08:13:41

Press CTRL-A Z for help on special keys                                                         
                                                                                                
I (1256) network_driver: AP mode configured                                                     
I (1256) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10                                 
W (1256) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
W (1296) phy_init: saving new calibration data because of checksum failure, mode(2)
I (1306) wifi:mode : softAP (24:58:7c:5a:f5:29)
I (1316) wifi:Total power save buffer number: 16
I (1316) wifi:Init max length of beacon: 752/752
I (1316) wifi:Init max length of beacon: 752/752
I (1316) network_driver: WIFI started
I (1316) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1326) network_driver: WIFI_EVENT_AP_START received.
I (1326) network_driver: AP DHCP hostname set to atomvm-24587c5af528
WLAN AP ready. Waiting connections.
Web server listening on port 8080
ALISP console listening on port 2323
Waiting to accept shell connection...
I (10696) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (10696) wifi:station: f8:ff:c2:11:4c:dd join, AID=1, bgn, 20
I (10716) network_driver: WIFI_EVENT_AP_STACONNECTED received.
I (10836) wifi:<ba-add>idx:2 (ifx:1, f8:ff:c2:11:4c:dd), tid:6, ssn:0, winSize:64
I (13116) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (13116) network_driver: IP_EVENT_AP_STAIPASSIGNED: 192.168.4.2
I (14906) wifi:<ba-add>idx:3 (ifx:1, f8:ff:c2:11:4c:dd), tid:0, ssn:2, winSize:64
CRASH 
======
pid: <0.12.0>

Stacktrace:
[{http_server,parse_query_string,1,[{file,"/__w/avm_tag_ci/avm_tag_ci/libs/eavmlib/src/http_server.erl"},{line,126}]},{http_server,loop,2,[{file,"/__w]

cp: #CP<module: 5, label: 84, offset: 42>

x[0]: error
x[1]: function_clause
x[2]: {2,2,59,1,[{12,388},{12,978}],error}

Stack 
------

[]
[{socket,{$avm_gen_tcp,<0.13.0>,gen_tcp_inet}},{header,"Accept-Language: en-US,en-GB;q=0.9,en;q=0.8"},{header,"Content-Length: 32"},{header,"Referer: ]
#CP<module: 12, label: 14, offset: 116>
#CP<module: 12, label: 103, offset: 0>

Mailbox
--------

Monitors
--------

**End Of Crash Report**
```

pguyot, 2024-10-13:
I confirm it builds with my esp-idf component using C++ (atomvm_m5) with 5.2 targetting esp32s3.

UncleGrumpy, 2024-10-13:
http(s) clients are working on esp32-s3, but don't seem to be working on esp32-c6.

UncleGrumpy, 2024-10-13:
`esp:deep_sleep(10000)` throws `badarg`

bettio, 2024-10-13:
> I don't seem to be able to boot esp32s2 regular (non-elixir) image. Elixir image looks ok (Wifi shows up and I can connect to HTTP and alisp servers).
>
> For the record, ESP32S2 (at least the board I [...]

This issue is really weird. `boot.avm` partition is not truncated and it is correctly placed. Also the factory partition looks correct.

bettio, 2024-10-13:
`AtomVM-linux-x86_64-static-mbedtls` works on ubuntu 22.04, it can start a complex Elixir application making using of atomgl plugin.

bettio, 2024-10-13:
Test network modules with `AtomVM-linux-x86_64-static-mbedtls` on Ubuntu 22.04.
`ssl`, `gen_tcp`, `gen_udp`, `socket` (both `udp` and `tcp`, both client and server) and `ahttp_client` using examples in `examples/erlang/`.

pguyot, 2024-10-13:
On second attempt, I've been able to boot S2 with regular image and connect to wifi and connect to lisp repl.