atomvm/AtomVM #1318: Take a decision on devmode for ESP32-H2

bettio, 2024-10-12:
According to tests devmode on H2 is not properly working.
There is no wifi, so it crashes when devmode starts.

petermm, 2024-10-12:
and presumably same for p4, I suggest a board check here https://github.com/atomvm/AtomVM/blob/c7e0943e64cf5f7662170fd2a971a6c3bd6ced37/libs/esp32devmode/src/esp32devmode.erl#L85 for those two boards.

UncleGrumpy, 2024-10-12:
Good point about the P4. I hope to work on Ethernet support soon, so P4 could still use devmode eventually.

UncleGrumpy, 2025-01-19:
Since disterl is on the horizon, and @pguyot had mentioned previously that disterl was not necessarily restricted to the network (i.e. could take place over UART) it might be interesting for devices that lack network peripherals to start as a disterl over UART node as an alternative devmode.