atomvm/AtomVM #1324: i2c:write_bytes panic on ESP32

ray-delossantos, 2024-10-13:
Using an ESP32 board

When running this function

```erlang

i2c_write_bytes_with_result(I2C, Bytes) -> 
    case i2c:write_bytes(I2C, <<0, 128>>) of
        ok -> {ok, nil};
        error -> {error, nil};
        {error, _} = E -> E
    end.

```

It throws the error below. 

```shell
# sudo cu -s 115200 -l /dev/tty.usbserial-0001

I (969) i2c_driver: I2C driver installed using I2C port 0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400da301  PS      : 0x00060030  A0      : 0x801395a3  A1      : 0x3ffba930
A2      : 0x00000002  A3      : 0x3ffcc16c  A4      : 0x3ffcc21c  A5      : 0x00000064
A6      : 0x03ffcc67  A7      : 0x3ffcc300  A8      : 0xfffffffc  A9      : 0x00000003
A10     : 0x00000033  A11     : 0x0000006c  A12     : 0x3ffbbd18  A13     : 0x00000167
A14     : 0x0000000b  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000068  LBEG    : 0x4008336d  LEND    : 0x40083375  LCOUNT  : 0x00000027

Backtrace: 0x400da2fe:0x3ffba930 0x401395a0:0x3ffba990 0x401396b6:0x3ffba9c0 0x400f3977:0x3ffba9e0 0x4011485a:0x3ffbaa90 0x400d90e7:0x3ffbaab0 0x401dca17:0x3ffbaaf0

```

What I may be doing wrong? 

Note: If I use the ```i2c:write_byte``` and send the exact same bytes one by one it works as expected.

bettio, 2024-10-15:
Thanks for reporting this issue, we need to reproduce first, and investigate this.
Since we were close to a release, we left this as a known issue and it is a good candidate for a v0.6.6 maintenance release.

ray-delossantos, 2024-10-16:
Thanks for the update. If it helps I may be able to test that part of the release when ready.

commit a73c2684 (2025-02-09): Merge pull request #1514 from UncleGrumpy/fix_i2c_write_bytes

Fix bug in i2c:write_bytes/2

Fixes a hard crash of the VM when `i2c:write_bytes/2` is used.

Improves logged errors in the i2c driver by displaying the decoded error string
rather than error number in console logs.

Closes #1324

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit e9c54196 (2025-02-09): Fix i2c:write_bytes on ESP32

Fixes a bug where input data was incorrectly processed befor being used in the ESP-IDF function
`i2c_master_write` leading to a hard crash of the VM.

Closes #1324
PR #1514 (2025-02-09): Fix bug in i2c:write_bytes/2
Fixes a hard crash of the VM when `i2c:write_bytes/2` is used.

Improves logged errors in the i2c driver by displaying the decoded error string rather than error number in console logs.

Closes #1324

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later