atomvm/AtomVM #1343: Help needed with Pi Pico

kevinschweikert, 2024-10-15:
Hey, really loving this project and trying get my feet wet. I'm having troubles getting my program to run on the Pi Pico.

I installed the two `.uf2` files with:

```bash
cp ~/Downloads/AtomVM-pico-v0.6.5.uf2 /Volumes/RPI-RP2/.
cp ~/Downloads/atomvmlib-v0.6.5.uf2 /Volumes/RPI-RP2/.
mix compile
mix atomvm.packbeam
cp atom_experiment.avm /Volumes/RPI-RP2/.
```

My `mix.exs` looks like:

```elixir
defmodule AtomExperiment.MixProject do
  use Mix.Project

  def project do
    [
      app: :atom_experiment,
      version: "0.1.0",
      elixir: "~> 1.17",
      start_permanent: Mix.env() == :prod,
      deps: deps(),
      atomvm: atomvm()
    ]
  end

  # Run "mix help compile.app" to learn about applications.
  def application do
    [
      extra_applications: [:logger]
    ]
  end

  # Run "mix help deps" to learn about dependencies.
  defp deps do
    [
      # {:dep_from_hexpm, "~> 0.3.0"},
      # {:dep_from_git, git: "https://github.com/elixir-lang/my_dep.git", tag: "0.1.0"}
      {:exatomvm, git: "https://github.com/atomvm/ExAtomVM/"}
    ]
  end

  defp atomvm do
    [
      start: AtomExperiment,
      flash_offset: 0x210000
    ]
  end
end
```

And the start module looks like this:

```elixir
defmodule AtomExperiment do
  def start do
    spawn(fn -> hello(0) end)
    Process.sleep(:infinity)
  end

  def hello(counter) do
    :io.format(~c"Hello No ~p\n", [counter])
    Process.sleep(1_000)
    hello(counter + 1)
  end
end
```

When i run the generated `.avm` file with

```bash
> atomvm atom_experiment.avm 
Hello No 0
Hello No 1
^C⏎      
``` 

it works fine. But in the minicom console i get this:

```
Welcome to minicom 2.9

OPTIONS: 
Compiled on Sep 22 2023, 21:10:41.
Port /dev/tty.usbmodem111201, 16:38:02

Press Meta-Z for help on special keys

    ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ## 
      ## ##      ##    ##     ## ###   ### ##     ## ###   ### 
     ##   ##     ##    ##     ## #### #### ##     ## #### #### 
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ## 
    #########    ##    ##     ## ##     ##  ##   ##  ##     ## 
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ## 
    ##     ##    ##     #######  ##     ##    ###    ##     ## 

    ###########################################################

Starting AtomVM revision 0.6.5
Fatal error: invalid main.avm packbeam
Aborted
AtomVM application terminated.  Going to sleep forever ...
```      

How can i debug the issue? Would really appreciate if you could help me out.

kevinschweikert, 2024-10-15:
I made it work with `mix atomvm.pico.flash`... maybe the `flash_offset` was the issue and it got overwritten/ignored by the mix task?

UncleGrumpy, 2024-10-15:
The flash_offset is only used by esp32, this setting is ignored when using the pico. The mix plugin was just updated for pico support and is the recommended method when working in Elixir.

So just to confirm, it does work as expected using the atomvm.pico.flash task?

UncleGrumpy, 2024-10-15:
I did forget that flash_offset is also used by the  STM32 platform. Regardless, it is ignored by pico devices.

kevinschweikert, 2024-10-15:
Yes, it does work with `mix atomvm.pico.flash`!

UncleGrumpy, 2024-10-15:
> Yes, it does work with `mix atomvm.pico.flash`!

Fantastic. Unless you need further help, would you mind closing this issue? Of course feel free to open future issues for any specific problems you encounter, or places where our documentation is confusing.

closed 2024-10-16 by kevinschweikert