atomvm/AtomVM #1350: external format doesn't support pids

pguyot, 2024-10-22:
Instead of a badarg exception, when calling `term_to_binary(self()).`, VM prints:
```
Unknown external term type: 19
```
and then aborts.

pguyot, 2024-11-03:
Fixed in #1358

UncleGrumpy, 2025-01-20:
Closing this because the fix #1358 was merged. It was no closed automatically because the PR comment said "Fixes" rather than "Closes" #1350 ;-)

closed 2025-01-20 by UncleGrumpy

pguyot, 2025-01-21:
> Closing this because the fix #1358 was merged. It was not closed automatically because the PR comment said "Fixes" rather than "Closes" #1350 ;-)

I believe it wasn't closed because commit from PR 1358 was merged on a branch and not on main.

UncleGrumpy, 2025-01-21:
Oh shoot! You are right, this isn't in main yet. I thought the auto-close feature was just picky about the using "closes".

commit fae8f601 (2024-12-21): Merge pull request #1358 from pguyot/w44/encode_and_decode_local_pids

Add support for encoded local pids in external terms

Fixes #1350

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #1358 (2024-12-21): Add support for encoded local pids in external terms
Fixes #1350 

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later