atomvm/AtomVM #1361: :esp.rtc_slow_get_binary() crash when mem unset

petermm, 2024-11-07:
if one calls `:esp.rtc_slow_get_binary()` immediately on start (eg. no rtc mem set) atomvm crashes:
```
AtomVM init.
I (651) sys: Loaded BEAM partition main.avm at address 0x250000 (size=1048576 bytes)
Starting application...
CRASH 
======
pid: <0.1.0>

Stacktrace:
Unable to allocate space for module paths.  Returning raw stacktrace.
{1,0,0,0,[{4,367}],error}

cp: #CP<module: 0, label: 34, offset: 0>

x[0]: error
x[1]: badarg
x[2]: {1,0,0,0,[{4,367}],error}

Stack 
------

#CP<module: 0, label: 34, offset: 0>
```

functions works after setting the memory to something `:esp.rtc_slow_set_binary(<<"some binary data">>)`

NB: You may need to poweroff/unplug to unset rtc_slow_mem, and get back to the crash.

petermm, 2024-11-07:
ok, I'll make a doc PR.

UncleGrumpy, 2024-11-07:
That being said, it does make for slightly more complicated code on the users part. This a perfect case for the use of `maybe`.

petermm, 2024-11-07:
> That being said, it does make for slightly more complicated code on the users part. This a perfect case for the use of `maybe`.

Yeah, but in elixir land `with` doesn't handle a raise - so it's gets verbose.

But I should just have read the docs here: https://www.atomvm.net/doc/main/apidocs/erlang/eavmlib/esp.html#rtc-slow-get-binary-0

So PR with a small improvement to Programmers Guide, these are also just primitives, one should totally wrap them and serialize/deserialize etc etc.

closed 2024-11-07 by petermm