atomvm/AtomVM #1371: esp32: system_architecture returns <<"Generic--">>

petermm, 2024-11-20:
`erlang:system_info(system_architecture)`, seems off.

0.6.5 release image and local builds return `<<"Generic--">>`

code search suggests it should return something like `<<"ESP_IDF-3.2-xtensa_esp32">>`

UncleGrumpy, 2024-12-16:
So the documentation is not clear, or our implementation is wrong… this is returning the OS architecture as defined by the compiler… all of the MCU platforms will report “Generic” since they are using neither X86_64 or ARM compilers.

UncleGrumpy, 2024-12-20:
I should have said as defined by cmake configuration, not necessarily the compiler itself.

petermm, 2024-12-20:
Yeah,

I cooked this up (from some of your code):

**src/platforms/esp32/components/libatomvm/CMakeLists.txt**
```
if (CMAKE_SYSTEM_NAME STREQUAL "Generic")
     if (CONFIG_IDF_TARGET_ARCH_XTENSA)
         set(CMAKE_SYSTEM_NAME "xtensa")
     elseif(CONFIG_IDF_TARGET_ARCH_RISCV)
         set(CMAKE_SYSTEM_NAME "riscv32")
     endif()
 endif()

 if (CMAKE_SYSTEM_PROCESSOR STREQUAL "")
     if (CONFIG_IDF_TARGET)
         set(CMAKE_SYSTEM_PROCESSOR "${CONFIG_IDF_TARGET}")
     endif()
 endif()

 if (CMAKE_SYSTEM_VERSION STREQUAL "")
         execute_process(
             COMMAND idf.py --version
             WORKING_DIRECTORY ${CMAKE_SOURCE_DIR}
             OUTPUT_VARIABLE SYSTEM_VERSION
             OUTPUT_STRIP_TRAILING_WHITESPACE
         )
         # Use idf_get_version() to get the version directly
         string(REPLACE "ESP-IDF" "" SYSTEM_VERSION "${SYSTEM_VERSION}")
         string(REPLACE "-" "_" SYSTEM_VERSION "${SYSTEM_VERSION}")
         string(STRIP "${SYSTEM_VERSION}" SYSTEM_VERSION)
         string(REPLACE " " "" SYSTEM_VERSION "${SYSTEM_VERSION}")
     if (NOT SYSTEM_VERSION STREQUAL "")
         set(CMAKE_SYSTEM_VERSION "${SYSTEM_VERSION}")
     endif()
 endif()
```

which gives you: **"riscv32-v5.3.2_210_g12938e511e_dirty-esp32c3"**

but it runs into build issue due this "Generic" check
https://github.com/atomvm/AtomVM/blob/1ca523c6c9834e890ad59ac611703a9d3ea4f048/src/libAtomVM/CMakeLists.txt#L278

And I have a very limited understanding of the build system, especially across platforms.. 

but the utility of `erlang:system_info(system_architecture)`, so low priority, and mostly it's deciding what output should be.

UncleGrumpy, 2024-12-20:
I don't think we want to chant the CMAKE_ configuration there, that would lead to build problems... but I believe we could add a separate definition for the compiler.

UncleGrumpy, 2024-12-20:
That definition for Generic is important for all of the MCU platforms in the cmake configuration process, so we don't want to mess with that.

UncleGrumpy, 2024-12-28:
This is also slightly related to PR #1117 which I closed. I didn’t feel like we ever came to a logical consensus on what the returns should be, and as you point out, is  not frequently needed.