atomvm/AtomVM #1377: Possible bug in http_server routing

danifv, 2024-11-25:
Hello,
I am trying to implement a http API, and found a possible issue in how routing is performed.
In
https://github.com/atomvm/AtomVM/blob/682369776d39895cdaeb101846dfa811c74dca99/libs/eavmlib/src/http_server.erl#L52-L55
I found that a matching `Module` is found based on whether I have a `Route` with `Target` of an equal value.  This impedes using query strings, and possibly other use cases as well.
Hopefully I did not miss anything. Could you please take a look at it?

pguyot, 2024-11-25:
Built-in http_server is very basic and I understand your frustration. You can use "*" as a target and have a module that will process URLs with query strings.

Alternatively, you can use httpd in atomvm_lib which has more features.
https://github.com/atomvm/atomvm_lib/blob/master/src/httpd.erl

danifv, 2024-11-30:
Thank you for the note!  
I was not up-to-date with all the features in `atomvm_lib`.

closed 2024-11-30 by danifv