atomvm/AtomVM #1379: VM heap buffer overflow crash

jakub-gonet, 2024-11-26:
When compiling with ASAN (I added `-g` but didn't provide line numbers) it reports heap overflow (logs below).
Is this a known error?

Compiled with Erlang/OTP 26 [erts-14.2.5.4] on M3 Pro, arm64.
```
target_compile_options(libAtomVM PUBLIC -fsanitize=address -fno-omit-frame-pointer)
target_link_options(libAtomVM PUBLIC -fsanitize=address)
```
Crashing code:
```erl
-module(hello_world).
-export([start/0]).
-export([f/0]).

start() ->
    ?MODULE:f().

f() ->
    [
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0},
        {f, fun f/0}
    ].
```

<details><summary>Logs</summary>
<p>

```
make_fun3/3, fun_index=0 dreg=x14 numfree=0
-- loop -- i = 112, next ocopde = 171
make_fun3/3, fun_index=0 dreg=x15 numfree=0
-- loop -- i = 117, next ocopde = 171
make_fun3/3, fun_index=0 dreg=y-959 numfree=0
-- loop -- i = 123, next ocopde = 164
put_tuple2/2, size=2
=================================================================
==35403==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x61a000003b90 at pc 0x000100f5aff0 bp 0x00016efb3c10 sp 0x00016efb3c08
WRITE of size 8 at 0x61a000003b90 thread T0
    #0 0x100f5afec in scheduler_entry_point+0x124c4c (AtomVM:arm64+0x10013afec)
    #1 0x100e3634c in context_execute_loop+0x50c (AtomVM:arm64+0x10001634c)
    #2 0x100e2329c in main+0xf68 (AtomVM:arm64+0x10000329c)
    #3 0x18b053150 in start+0x9a8 (dyld:arm64e+0xfffffffffff4d150)

0x61a000003b90 is located 0 bytes after 1296-byte region [0x61a000003680,0x61a000003b90)
allocated by thread T0 here:
    #0 0x101cacc04 in malloc+0x94 (libclang_rt.asan_osx_dynamic.dylib:arm64e+0x54c04)
    #1 0x100fc8110 in memory_init_heap+0x20 (AtomVM:arm64+0x1001a8110)
    #2 0x100fc9d88 in memory_gc+0x26c (AtomVM:arm64+0x1001a9d88)
    #3 0x100fc9030 in memory_ensure_free_with_roots+0x72c (AtomVM:arm64+0x1001a9030)
    #4 0x100e4878c in scheduler_entry_point+0x123ec (AtomVM:arm64+0x10002878c)
    #5 0x100e3634c in context_execute_loop+0x50c (AtomVM:arm64+0x10001634c)
    #6 0x100e2329c in main+0xf68 (AtomVM:arm64+0x10000329c)
    #7 0x18b053150 in start+0x9a8 (dyld:arm64e+0xfffffffffff4d150)

SUMMARY: AddressSanitizer: heap-buffer-overflow (AtomVM:arm64+0x10013afec) in scheduler_entry_point+0x124c4c
Shadow bytes around the buggy address:
  0x61a000003900: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x61a000003980: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x61a000003a00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x61a000003a80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x61a000003b00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x61a000003b80: 00 00[fa]fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x61a000003c00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x61a000003c80: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x61a000003d00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x61a000003d80: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x61a000003e00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==35403==ABORTING
[1]    35403 abort      ../../src/AtomVM parse_forms.avm
```

</p>
</details>

pguyot, 2024-11-26:
I confirm the crash. I believe there is a bug in the xregs > 15 implementation. The crash doesn't happen with the implementation in PRs #953 or #954. They can be resurrected if needed.

bettio, 2024-11-29:
Problem found, a small fix will arrive soon in release-0.6. This may  also allow a small optimization too.

PR #1381 (2024-12-11): Fix WRITE_REGISTER_GC_SAFE memory corruption
Fix #1379

In few words, `((dreg_gc_safe).base == x_regs)` is not true for x registers >= 16.

Also there is a very minor optimization.

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit 89438f35 (2024-12-11): Merge pull request #1381 from bettio/fix-write-register-gc-safe

Fix WRITE_REGISTER_GC_SAFE memory corruption

Fix #1379

In few words, `((dreg_gc_safe).base == x_regs)` is not true for
x registers >= 16.

Also there is a very minor optimization.

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2024-12-13 by bettio