atomvm/AtomVM #1384: Possible memory leak when using linked processes

danifv, 2024-12-04:
I'm doing some measurements on resource usage (memory and power mainly) and found a possible memory leak.  
When spawning linked processes that end normally, the available Free Heap is lower than before starting the processes.  
e.g.
```
f() ->
    [spawn_link(fun() ->  timer:sleep(500) end) || _ <- lists:seq(1, 50)].
```

In contrast, when spawning linked processes that terminate, the available Free Heap is more or less the same as before starting the processes.
e.g.
```
f() ->
    [spawn(fun() ->  ephemeral_start() end) || _ <- lists:seq(1, 50)].

ephemeral_start() ->
    spawn_link(fun() -> ephemeral() end).

ephemeral() ->
    timer:sleep(500),
    exit(normal).
```

I suspect, that memory related to the link remains allocated.

I'm doing the measurements on an ESP32 system.

pguyot, 2024-12-05:
Is this related to this comment?

https://github.com/atomvm/AtomVM/blob/fc824ad569648e1b75b9055b8ef819503196923c/src/libAtomVM/context.c#L392

danifv, 2024-12-05:
It could possibly be if that code also relates to links.  
However I can confirm that a similar leak also occurs when `monitoring` spawned processes:
```
f() ->
    Pids = [spawn(fun() -> timer:sleep(500) end) || _ <- lists:seq(1,N)],
    [erlang:monitor(process, P) || P <- Pids].
```

pguyot, 2024-12-06:
Links are handled internally there, this is the case where `monitor->ref_ticks == 0`.

commit 33af0563 (2025-03-06): Merge pull request #1555 from pguyot/w08/fix-link-leak

Use signals for link/unlink/monitor/demonitor

Fix concurrency issues and memory leaks related to a poor usage of list of
monitors that was accessed by other processes (fix #1384)
Fix semantics of monitor where self cannot monitor itself and a monitor cannot
be removed from another process.

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit a639cbc6 (2025-03-06): Use signals for link/unlink/monitor/demonitor

Fix concurrency issues and memory leaks related to a poor usage of list of
monitors that was accessed by other processes (fix #1384)
Fix semantics of monitor where self cannot monitor itself and a monitor
cannot be removed from another process.
PR #1555 (2025-03-06): Use signals for link/unlink/monitor/demonitor
Fix concurrency issues and memory leaks related to a poor usage of list of monitors that was accessed by other processes (fix #1384)
Fix semantics of monitor where self cannot monitor itself and a monitor cannot be removed from another process.

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2025-03-08 by bettio