atomvm/AtomVM #1386: gpio:set_int(Gpio, Pin, none) -> error

petermm, 2024-12-06:
Somebody struggled and posted this in the Discord. Reading the docs, it seems this code should work

>  Available triggers are none (which is the same as disabling an interrupt), rising, falling, both (rising or falling), low, and high.

https://www.atomvm.net/doc/main/apidocs/erlang/eavmlib/gpio.html#set-int-3

> In my code I want to enable and disable interrupts but get an error when diabling the interrupts.
```erlang
start() ->
    start(undefined, undefined).

start(_StartType, _StartArgs) ->
    Int_pin = 33,
    Gpio = gpio:open(),
    gpio:set_direction(Gpio, Int_pin, input),
    ok = gpio:set_int(Gpio, Int_pin, rising),
    ok = gpio:set_int(Gpio, Int_pin, none),
    wait_forever().
```

> I get an error on the second gpio:set_int/3 call:

```erlang
CRASH
======
pid: <0.1.0>

Stacktrace:
[{mpu6050_app,start,2,[{file,"/home/username/mpu6050/src/mpu6050_app.erl"},{line,21}]}]

cp: #CP<module: 4, label: 4, offset: 64>

x[0]: error
x[1]: {badmatch,error}
x[2]: {1,1,39,1,[{4,105}],error}
```

Problem solved by using:
```erlang
ok = gpio:remove_int(Gpio, Int_pin)
```

But there may be a (doc?) issue.

UncleGrumpy, 2024-12-06:
This looks like a bug, I will try to look into this over the weekend. You already found the workaround. Thanks for reporting this!

UncleGrumpy, 2024-12-09:
I looked at our handling of gpio interrupts and we do not have a special case necessary to make `none` remove any interrupts set on the given pin. Since we have a dedicated function to disable the interrupts I think updating the docs makes more sense than making the code more complicated.

UncleGrumpy, 2024-12-16:
I thought about this a bit more, and after thinking of all the situations where using the `none` trigger would be preferable, I looked harder and found that it could be supported with minimal changes. See #1399.

commit a692149d (2024-12-17): Fix broken ESP32 gpio interrupt trigger `none`

Fixes the implementation of the `none` trigger that has been documented since release-0.5, but was
apparently never tested, its use causes a error (see #1386).

Moves the main functionality from `gpiodriver_remove_int` into a new `unregister_interrupt_listener`
funtion that is also used when setting the interrupt trigger to `none` to remove an interrupt.

Closes #1386
commit bf3783c2 (2024-12-17): Merge pull request #1399 from UncleGrumpy/esp32_gpio_int-none

Fix broken ESP32 gpio interrupt trigger `none`

Fixes the implementation of the `none` trigger that has been documented since
release-0.5, but was apparently never tested, its use causes a error (see #1386).

Moves the main functionality from `gpiodriver_remove_int` into a new
`unregister_interrupt_listener` funtion that is also used when setting the
interrupt trigger to `none` to remove an interrupt.

Closes #1386

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license t [... cut]
PR #1399 (2024-12-17): Fix broken ESP32 gpio interrupt trigger `none`
Fixes the implementation of the `none` trigger that has been documented since release-0.5, but was apparently never tested, its use causes a error (see #1386).

Moves the main functionality from `gpiodriver_remove_int` into a new `unregister_interrupt_listener` funtion that is also used when setting the interrupt trigger to `none` to remove an interrupt.

Closes #1386

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGP [... cut]
closed 2024-12-20 by petermm