atomvm/AtomVM #1388: 'unsupported line_ref tag: 0' warning when decoding line refs

mat-hek, 2024-12-06:
Hey there,

sometimes when there's an error raised I get the `unsupported line_ref tag: 0` warning and it seems it's correlated with missing entries in the stack trace. I'm going to debug further, but it seems that the warning comes from here: https://github.com/atomvm/AtomVM/blob/fc824ad569648e1b75b9055b8ef819503196923c/src/libAtomVM/module.c#L498-L499 and it seems a part of 'compact integers decoding' implementation is missing. Is there any place (in OTP?) where I can find some details / a reference implementation of it so that I can add it to AtomVM?

pguyot, 2024-12-06:
Thank you for rising this, it is indeed annoying 😀 

I'm not aware of any documentation and to handle similar cases in the past I ended up looking at BEAM source code, the VM or the compiler.

pguyot, 2024-12-06:
Well, I'd search the code in BEAM that builds the line table.
https://github.com/erlang/otp/blob/50c08864b6decbe3561143986bda06663a9da0b7/lib/compiler/src/beam_asm.erl#L295

And more precisely in encode_line_items :
https://github.com/erlang/otp/blob/50c08864b6decbe3561143986bda06663a9da0b7/lib/compiler/src/beam_asm.erl#L324

Reading the code, I don't understand our decoding and I wonder if it's correct as we do the same thing for `TAG_COMPACT_INT` and `TAG_COMPACT_ATOM` but we don't do the same thing for `TAG_EXTENDED_INT` and `TAG_EXTENDED_ATOM`.

There seems to be four way to encode the value depending, and we only do two. See:
https://github.com/erlang/otp/blob/50c08864b6decbe3561143986bda06663a9da0b7/lib/compiler/src/beam_asm.erl#L579

PR #1556 (2025-03-08): Fix handling of filenames in stack traces
- Fix a crash that happened if a stack element was in another file
- Optimize memory usage by not building lists of line references and of filenames
- Fix parsing of files with no filename at all.

This fixes most occurrences of 'unsupported line_ref tag: 0' (fixes #1388)

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit 59a54b84 (2025-03-08): Fix handling of filenames in stack traces

- Fix a crash that happened if a stack element was in another file
- Optimize memory usage by not building lists of line references and of filenames
- Fix parsing of files with no filename at all.

This fixes most occurrences of 'unsupported line_ref tag: 0' (fixes #1388)
commit bc21d17e (2025-03-08): Merge pull request #1556 from pguyot/w10/fix-stack-traces-filenames

Fix handling of filenames in stack traces

- Fix a crash that happened if a stack element was in another file
- Optimize memory usage by not building lists of line references and of filenames
- Fix parsing of files with no filename at all.

This fixes most occurrences of 'unsupported line_ref tag: 0' (fixes #1388)

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2025-03-08 by bettio