atomvm/AtomVM #1463: Support for bignums?

jakub-gonet, 2025-01-08:
I couldn't find the limitations section in the docs.
1. Is bignum support considered?
2. Would you accept a patch for `externalterm` to have better error reporting when it can't decode serialized data?

AVM communicate well enough if value is out of bounds in runtime but does pretty poor error reporting if loaded BEAM file has it as a literal:
https://github.com/atomvm/AtomVM/blob/23b078133a516e871bf936f8be6cfa61b5495291/src/libAtomVM/externalterm.c#L700-L702

This aborts the VM  (`module_load_literal`).

Example code:
```erlang
-module(hello_world).
-export([
    start/0,
    is_bignum/1
]).

start() ->
    erlang:display(?MODULE:is_bignum(1)),
    ok.

% crashes the VM
is_bignum(N) ->
    if
        N >= 1 bsl 128 ->
            big;
        N =< -(1 bsl 128) ->
            smol;
        true ->
            ok
    end.

%% dumps VM state
% is_bignum(N) ->
%     Big = rt_bsl(1, 128),
%     Smol = -rt_bsl(1, 128),
%     if
%         N >= Big ->
%             big;
%         N =< Smol ->
%             smol;
%         true ->
%             ok
%     end.

% rt_bsl(N, 0) -> N;
% rt_bsl(N, Shift) -> rt_bsl(2 * N, Shift - 1).
```

jakub-gonet, 2025-01-08:
Actually, duplicate of #1345.

closed 2025-01-08 by jakub-gonet

UncleGrumpy, 2025-01-08:
> I couldn't find the limitations section in the docs.
> 
> 1. Is bignum support considered?
> 

This should probably be up front in the introductory section of the docs, but for reference, bignums are covered in the "Limitations" section of the Programmer's Guide:

https://www.atomvm.net/doc/main/programmers-guide.html#limitations

jakub-gonet, 2025-01-08:
I'll update the other comment. Thanks!