atomvm/AtomVM #1479: Tests.ex: 'Invalid filename' when adding test modules

petermm, 2025-01-19:
For more elixir testing we need to use various test modules in Tests.ex.

_TIL sub/nested modules doesn't work due using elixirc for compilation._

In https://github.com/atomvm/AtomVM/pull/1476 I include them in exavmlib (they will go in own testexavmlib, but to minimise moving parts while resolving this issue, they are in exavmlib).

This leads to these warnings: (failing to parse filename?)
```
Invalid filename: expected filename data (27008 bytes).
Invalid filename: expected filename data (27008 bytes).
Invalid filename: expected filename data (27008 bytes).
Invalid filename: expected filename data (18719 bytes).
```

Tests/runtime still works, however when crashing with a stacktrace, it will segfault due to filenames having been set to NULL.

Any guidance?

UncleGrumpy, 2025-01-19:
If you look in tests/libs directory there are tests bundles to test each library. I would ad a exavmlib directory there, and pack the tests with the atomvmlib libraries. Then kust make sure the CMakeLists.txt packs the standard + exavmlib with the test, see here:
https://github.com/atomvm/AtomVM/blob/a00098a09055ef5751e409bedc18381a6f7f66e4/tests/libs/eavmlib/CMakeLists.txt#L35

petermm, 2025-01-19:
Yeah, but this is already done here for Tests.ex (it's the initial elixir testing) https://github.com/atomvm/AtomVM/blob/a00098a09055ef5751e409bedc18381a6f7f66e4/tests/libs/exavmlib/CMakeLists.txt

but something isn't quite right.. hopefully some oversight by me..

UncleGrumpy, 2025-01-20:
> Yeah, but this is already done here for Tests.ex (it's the initial elixir testing) https://github.com/atomvm/AtomVM/blob/a00098a09055ef5751e409bedc18381a6f7f66e4/tests/libs/exavmlib/CMakeLists.txt
> 
> but something isn't quite right.. hopefully some oversight by me..

You should be adding you test modules here in this file too, like the eavmlib tests:
https://github.com/atomvm/AtomVM/blob/a00098a09055ef5751e409bedc18381a6f7f66e4/tests/libs/eavmlib/CMakeLists.txt#L25

UncleGrumpy, 2025-01-20:
Now that I look a little harder it seems it is only missing: `pack_archive(Tests)`.

UncleGrumpy, 2025-01-20:
No, never mind that previous comment, I didn't realize all of the tests were in a single module called `Tests`.

UncleGrumpy, 2025-01-20:
So did you try putting `GenServerTest.CustomStack.ex`, and  `GenServerTest.Stack.ex` into `tests/libs/examvlib/` and adding them to a packed archive with `Tests.ex` (like eavmlib does with the set() and pack_archinve()), and add both the `MODULE:start()` to Tests.ex?

petermm, 2025-01-20:
I've tried it all, but it does seem like I should try and port `packtest` macro to the elixir side and try to replicate that tests.erl pattern in Tests.ex.

UncleGrumpy, 2025-01-20:
> I've tried it all, but it does seem like I should try and port `packtest` macro to the elixir side and try to replicate that tests.erl pattern in Tests.ex.

Yes! I believe this is it. I started comparing the differences and this it the same conclusion I came to.

petermm, 2025-01-21:
the bug is in the nested/sub modules, eg. `GenServerTest.CustomStack` will give the error, but identical module/code within `GenServerTest` will give no error.

petermm, 2025-04-08:
This issue is no longer present, and the issue of adding Elixir modules is handled by: https://github.com/atomvm/AtomVM/pull/1625

closed 2025-04-08 by petermm