atomvm/AtomVM #1513: term_to_binary doesn't support lambdas

jakub-gonet, 2025-02-06:
Following code:
```erl
-module(hello_world).
-export([f/0, start/0]).

f() ->
    term_to_binary(fun() -> ok end).

start() ->
    ?MODULE:f(),
    ok.
```

crashes with
```
Unknown external term type: 106583908418080
```

Using external functions (`term_to_binary(fun ?MODULE:f/0)`) works.

pguyot, 2025-02-08:
Question is, should we support this if `erl_eval` is not in the path?

jakub-gonet, 2025-02-10:
I don't get the question, are you asking if anything else than code in `erl_eval` needs that? The code bloat is a good argument but IMO it should be discussed more generally as well, sometimes you don't e.g. need all ETS machinery.

FWIW, we also need `compressed`, `deterministic`, and `{minor_version, 2}` for our usecase. I wonder if supporting all terms is needed for distribution as well.

elsbiet, 2025-03-15:
and it would also be nice if binary_to_term(Bin) would work when Bin is the binary representation of a fun as this potentially would enable me to implement some cli functionality.