atomvm/AtomVM #1533: `test_estdlib.avm` segfault

bettio, 2025-02-11:
```
Run ulimit -c unlimited
I otp_socket: Peer closed connection. (/home/runner/work/AtomVM/AtomVM/src/libAtomVM/otp_socket.c:1936)
I otp_socket: Peer closed connection. (/home/runner/work/AtomVM/AtomVM/src/libAtomVM/otp_socket.c:1936)
/home/runner/work/_temp/941aafb2-76d3-4216-ab1a-b9ee80939e0e.sh: line 2: 14972 Segmentation fault      (core dumped) ./src/AtomVM ./tests/libs/estdlib/test_estdlib.avm
+
Error: Process completed with exit code 139.
```

See also: https://github.com/atomvm/AtomVM/actions/runs/13269224177/job/37044417655

## Core dump
```
SHA256 hash of uploaded artifact zip is 4d6332042a8070bf347773896b36274f27a952341be27007e4f478d4455331ca
Finalizing artifact upload
Artifact core-ubuntu-20.04-clang-10-26-13269224177-1.zip successfully finalized. Artifact ID 2573420704
Artifact core-ubuntu-20.04-clang-10-26-13269224177-1 has been successfully uploaded! Final size is 1500209 bytes. Artifact ID is 2573420704
Artifact download URL: https://github.com/atomvm/AtomVM/actions/runs/13269224177/artifacts/2573420704
``
[core-ubuntu-20.04-clang-10-26-13269224177-1.zip](https://github.com/user-attachments/files/18757389/core-ubuntu-20.04-clang-10-26-13269224177-1.zip)

UncleGrumpy, 2025-02-11:
I can't find too much to go in here, a debug build with symbols included would probably have a lot more useful information.
```
(gdb) where full
#0  0x0000000000452d8a in context_destroy ()
No symbol table info available.
#1  0x000000000040a4ea in scheduler_entry_point ()
No symbol table info available.
#2  0x000000000048256f in scheduler_thread_entry_point ()
No symbol table info available.
#3  0x00007f85cbc68609 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
No symbol table info available.
#4  0x0000000000000000 in ?? ()
No symbol table info available.
```

I also get some warnings about mismatched library versions, a statically compiled version of AtomVM would be much better for debugging on a different machine.
```
warning: .dynamic section for "/lib/x86_64-linux-gnu/librt.so.1" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libm.so.6" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libz.so.1" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libdl.so.2" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libpthread.so.0" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib/x86_64-linux-gnu/libc.so.6" is not at the expected address (wrong library or version mismatch?)

warning: .dynamic section for "/lib64/ld-linux-x86-64.so.2" is not at the expected address (wrong library or version mismatch?)
```

UncleGrumpy, 2025-02-11:
I am using gdb on Ubuntu 24.04, but making sure the installed library version matches the version used during the workflow run is very difficult.

pguyot, 2025-02-11:
I'm pretty sure I know where this happen, as the same crash happened on ubuntu 24.04 where the stacktrace appears in CI logs. It's a problem in the ref count of resources.

I did fix something in #1498 but crashes still occurred. I have yet to fix the tests with emscripten.

UncleGrumpy, 2025-02-11:
I just realized that workflow was run on 20.04, that explains the version errors, but regardless if these were statically linked, debug builds we would have a lot more information to work with.