atomvm/AtomVM #1551: ahttp_client:connect is taking up 40kb of memory

robinchew, 2025-02-19:
Just running the following piece of code will take up about 40,000 bytes of memory.
```erlang
Conn = case ahttp_client:connect(https, "mysensor.blob.core.windows.net", 443, [
    {active, false}, {verify, verify_none}, {parse_headers, [<<"Location">>]}
]) of
    {ok, C} -> C;
        E ->  print("Azure connection error ~p~n", [E])
    end,
```

robinchew, 2025-02-19:
Note that this memory exhaustion seems to only happen on `gassensor.blob.core.windows.net`. If you try on `mysensor.blob.core.windows.net`, it doesn't seem to exibit this problem.

petermm, 2025-02-19:
related in sdkconfig:
```
#
# mbedTLS
#
CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=y
# CONFIG_MBEDTLS_DEFAULT_MEM_ALLOC is not set
# CONFIG_MBEDTLS_CUSTOM_MEM_ALLOC is not set
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=16384
CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN=4096
# CONFIG_MBEDTLS_DYNAMIC_BUFFER is not set
# CONFIG_MBEDTLS_DEBUG is not set
```

and see this table from https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mbedtls.html#reducing-heap-usage

So it's somewhat expected I think.. So probably a docs issue, and test using different configs..

petermm, 2025-02-19:
MBEDTLS_ECP_FIXED_POINT_OPTIM was recently default disabled in esp-idf.

```
   This configuration option enables optimizations to speedup (about 3 ~ 4 times) the ECP
   fixed point multiplication using pre-computed tables in the flash memory.
   Disabling this configuration option saves flash footprint (about 29KB if all Elliptic Curve selected)
   in the application binary.
```

There are some suggestions that enabling it will reduce peak memory usage, at the cost of ~29kb of flash.