atomvm/AtomVM #1560: Memory corruption while running ESP32 `test_wifi_example`

bettio, 2025-03-10:
2025-03-10 20:58:15 Running test_wifi_example...
2025-03-10 20:58:15 I (14168) network_driver: Initialized network interface
2025-03-10 20:58:15 I (14168) network_driver: Created default event loop
2025-03-10 20:58:15 I (14168) AtomVM: Running start/0 from test_wifi_example.beam...
2025-03-10 20:58:15 
2025-03-10 20:58:15 E (14178) network_driver: Device is not connected to any AP.
2025-03-10 20:58:15 I (14178) network_driver: STA ssid: Wokwi-GUEST
2025-03-10 20:58:15 W (14178) network_driver: Warning: Empty password.  AP will be open network!
2025-03-10 20:58:15 bug: found unknown term type: 0x2B
2025-03-10 20:58:15 
2025-03-10 20:58:15 abort() was called at PC 0x40123c77 on core 0
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 Backtrace: 0x4008181a:0x3ffbabd0 0x4008c651:0x3ffbabf0 0x40094062:0x3ffbac10 0x40123c77:0x3ffbac80 0x401242c6:0x3ffbacd0 0x40124541:0x3ffbad10 0x4013dbcd:0x3ffbad30 0x400f3c21:0x3ffbad50 0x4012062a:0x3ffbae10 0x400da2ae:0x3ffbae30 0x400da5aa:0x3ffbae50 0x400e60c2:0x3ffbae70 0x400e6199:0x3ffbaea0 0x400e624c:0x3ffbafc0 0x400da681:0x3ffbafe0 0x401f0c27:0x3ffbb000
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 ELF file SHA256: e381d59e2
2025-03-10 20:58:16 
2025-03-10 20:58:16 
2025-03-10 20:58:16 Rebooting...

bettio, 2025-03-10:
Let's check if this crash is related in any manner to the previous error:
```
2025-03-10 20:58:38 I (27788) AtomVM: Running start/0 from test_rtc_slow.beam...
2025-03-10 20:58:38 
2025-03-10 20:58:38 CRASH 
2025-03-10 20:58:38 ======
2025-03-10 20:58:38 pid: <0.1.0>
2025-03-10 20:58:38 
2025-03-10 20:58:38 Stacktrace:
2025-03-10 20:58:38 
2025-03-10 20:58:38 undefined
2025-03-10 20:58:38 
2025-03-10 20:58:38 cp: #CP<module: 0, label: 27, offset: 0>
2025-03-10 20:58:38 
2025-03-10 20:58:38 x[0]: error
2025-03-10 20:58:38 x[1]: {badmatch,<<0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,43>>}
2025-03-10 20:58:39 x[2]: error
2025-03-10 20:58:39 
2025-03-10 20:58:39 Stack 
2025-03-10 20:58:39 
2025-03-10 20:58:39 -----
2025-03-10 20:58:39 
2025-03-10 20:58:39 []
2025-03-10 20:58:39 
2025-03-10 20:58:39 []
2025-03-10 20:58:39 #CP<module: 0, label: 27, offset: 0>
2025-03-10 20:58:39 
2025-03-10 20:58:39 
2025-03-10 20:58:39 
2025-03-10 20:58:39 
2025-03-10 20:58:39 Mailbox
2025-03-10 20:58:39 -------
2025-03-10 20:58:39 
2025-03-10 20:58:39 
2025-03-10 20:58:39 Monitors
2025-03-10 20:58:39 --------
2025-03-10 20:58:39 
2025-03-10 20:58:39 
2025-03-10 20:58:39 **End Of Crash Report**
2025-03-10 20:58:39 
2025-03-10 20:58:39 AtomVM finished with return value: error
2025-03-10 20:58:39 
2025-03-10 20:58:39 ./main/test_main.c:585:test_rtc_slow:FAIL:Function [test_run].  Expression Evaluated To FALSE
```

petermm, 2025-03-10:
this happens after  https://github.com/atomvm/AtomVM/pull/1546 - seems related to network_driver callbacks - reverting the PR change makes everything green..

I can replicate real esp32 and c3, real devices seems to be green if this network:sta_rssi(); is commented out, https://github.com/atomvm/AtomVM/blob/c9051f15e208873e57ff7e0b1d856158625b2d07/src/platforms/esp32/test/main/test_erl_sources/test_wifi_example.erl#L66

but CI is still bad with it commented out https://github.com/petermm/AtomVM/actions/runs/13444665723/job/37567165049

petermm, 2025-03-10:
> Let's check if this crash is related in any manner to the previous error:

timestamps;-)  this error is after the crashing starts, and only happens due to the restarts..

UncleGrumpy, 2025-03-11:
> Line 66 in [c9051f1](/atomvm/AtomVM/commit/c9051f15e208873e57ff7e0b1d856158625b2d07)
> 
>  network:sta_rssi(),

This should not be used in the sim test. What you get is the equivalent of an Ethernet interface, there is no wireless signal strength to be measured. Also this will always return an error if a station mode connection is not active. This only measures the signal strength to the associated access point.

UncleGrumpy, 2025-03-11:
My second point is still valid, using network:sta_rssi/0 before a connection is established will always result in an error. We should expect to catch an error here and only expect success after a connected callback.