atomvm/AtomVM #1561: ESP32 `test_rtc_slow` test fails

bettio, 2025-03-10:
Following test fails in [main](https://github.com/atomvm/AtomVM/commit/bafd18b7869e319165a92f2f91b1901811d07cc9):

```
2025-03-10 20:58:14 I (14068) AtomVM: Running start/0 from test_rtc_slow.beam...
2025-03-10 20:58:14 
2025-03-10 20:58:14 
2025-03-10 20:58:14 CRASH 
2025-03-10 20:58:14 ======
2025-03-10 20:58:14 pid: <0.1.0>
2025-03-10 20:58:14 
2025-03-10 20:58:14 Stacktrace:
2025-03-10 20:58:14 undefined
2025-03-10 20:58:14 
2025-03-10 20:58:14 cp: #CP<module: 0, label: 27, offset: 0>
2025-03-10 20:58:14 
2025-03-10 20:58:14 x[0]: error
2025-03-10 20:58:14 x[1]: {badmatch,<<0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,43>>}
2025-03-10 20:58:15 x[2]: error
2025-03-10 20:58:15 
2025-03-10 20:58:15 Stack 
2025-03-10 20:58:15 -----
2025-03-10 20:58:15 
2025-03-10 20:58:15 []
2025-03-10 20:58:15 []
2025-03-10 20:58:15 #CP<module: 0, label: 27, offset: 0>
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 Mailbox
2025-03-10 20:58:15 
2025-03-10 20:58:15 -------
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 Monitors
2025-03-10 20:58:15 --------
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 
2025-03-10 20:58:15 **End Of Crash Report**
2025-03-10 20:58:15 AtomVM finished with return value: error
2025-03-10 20:58:15 ./main/test_main.c:585:test_rtc_slow:FAIL:Function [test_run].  Expression Evaluated To FALSE
```

See also: https://github.com/atomvm/AtomVM/actions/runs/13771138211/job/38520707727#step:9:424

bettio, 2025-03-10:
Let's investigate if it is releated to #1560

petermm, 2025-03-10:
this happens due to a crash and restart(from other test), which means this line which is expected to be catch'ed and return ok, returns the actual rtc_slow https://github.com/atomvm/AtomVM/blob/bafd18b7869e319165a92f2f91b1901811d07cc9/src/platforms/esp32/test/main/test_erl_sources/test_rtc_slow.erl#L30