atomvm/AtomVM #1577: pico-w crash calling gen_tcp:connect/3

gpummer, 2025-03-15:
My erlang code crashes with a pico-w with AtomVM 0.6.4 or 0.6.5 (including corresponding atomvmlib-v0.6.*):
```
CRASH 
======
pid: <0.1.0>

Stacktrace:
[{gen_tcp_inet,connect,3,[{file,"/home/runner/work/AtomVM/AtomVM/libs/estdlib/src/gen_tcp_inet.erl"},{line,55}]},{gen_tcp,connect,3,[{]

cp: #CP<module: 9, label: 2, offset: 48>

x[0]: error
x[1]: badarg
x[2]: {3,3,208,3,[{0,105},{9,58},{10,31}],error}

Stack 
------

[{active,true}]
1701
{10,0,0,34}
#CP<module: 9, label: 2, offset: 48>
[]
[]
gen_tcp_inet
#CP<module: 0, label: 4, offset: 82>
[]
[]
#CP<module: 0, label: 22, offset: 0>
``` 

What I tried: Giving no options at all ([]), all options set ([{active, false}, {buffer, 1024}, {timeout, 20000}, listx]). I known that there is an host at the given address/port which responds. I tried also giving a hostname, a FQDN. But no success.

I tried to understand the source code of estdlib/gen_tcp(_inet).erl. I think that this has something to do with open_port({spawn, "socket"}, []), but I am too new in the erlang world.
 
What did I wrong? Or is this an issue with "pico-w can't connect via tcp / http_client because it's not implemented" yet? The same source code works with my esp32 device. 

I started with ahttp_client:connect but this also results in the same crash dump.

If it helps:
```erlang
 other_start() ->
    ok = maybe_start_network(atomvm:platform()),
    Host = maps:get(host, config:get()),
    Port = maps:get(port, config:get()),
    io:format("Attempting to connect to host ~p on port ~p ...~n", [Host, Port]),
    case gen_tcp:connect(Host, Port, [{active, true}]) of
        {ok, Socket} ->
            io:format("Connected!~n"),
            loop(Socket);
        Error ->
            io:format("An error occurred connecting: ~p~n", [Error])
    end.
``` 
maybe_start_network... is copied from the tcp_client sample.

Any hint is appreciated.

UncleGrumpy, 2025-03-15:
@petermm is correct about using binary for ssid and oak. You can took at the rp2040 tcp_server example for how to start the network:
https://github.com/atomvm/AtomVM/blob/main/examples/erlang/rp2040/picow_tcp_server.erl

pguyot, 2025-03-16:
The problem is inet backend is not available (and probably nevery will) on pico platform.
You should use the socket backend which is not the default yet. Please pass `{inet_backend, socket}` in the options to `gen_tcp:connect/3`. It should work on both pico and esp32.

gpummer, 2025-03-16:
> I think it should work, maybe post more of the code..
> 
> believe there is something about the ssid and password having to binary for picow eg. <<"ssid">> - see this example https://github.com/atomvm/AtomVM/blob/main/examples/erlang/rp2040/picow_wifi_sta.erl

I tried this. It seems to be unnecessary with newer versions. I tried v0.6.4 and v0.6.5 with "<<value>>" and "value". Both variants connect to the station and get an IP. Many thanks for your reply!

gpummer, 2025-03-16:
> The problem is inet backend is not available (and probably nevery will) on pico platform. You should use the socket backend which is not the default yet. Please pass `{inet_backend, socket}` in the options to `gen_tcp:connect/3`. It should work on both pico and esp32.

Many, many thanks! This is it!

Just for further reference / search:
```erlang
    case gen_tcp:connect(Host, Port, [{inet_backend, socket}]) of
        {ok, Socket} ->
            io:format("Connected!~n"),
```

I will close this issue and continuing my journey in the erlang and pico(2)-w world.

closed 2025-03-16 by gpummer