atomvm/AtomVM #1581: v0.6.6 release

bettio, 2025-03-16:
A lot of fixes has been merged into release-0.6, so it is time to release v0.6.6.
Before merging v0.6.6 following PRs should be merged into release-0.6:

- [ ] https://github.com/atomvm/AtomVM/pull/1564
- [ ] https://github.com/atomvm/AtomVM/pull/1569

After all PRs are merged extensive manual testing should be performed.

Test list is WIP.

UncleGrumpy, 2025-03-17:
It would be great if we could get #1499 merged before we cut a new release. It helps keep the "Production" GitHub Pages branch from growing unnecessarily large by not pushing unneeded build artifacts to the published pages, and removes old v0.6.0 pre-release branches from the menu. I already had to make those symlinks to the v0.6.0 release to keep the Production branch size down.

bettio, 2025-03-18:
> It would be great if we could get [#1499](https://github.com/atomvm/AtomVM/pull/1499) merged before we cut a new release. It helps keep the "Production" GitHub Pages branch from growing [...]

Agreed.

jgonet, 2025-05-22:
All linked PRs are merged now.