atomvm/AtomVM #1589: "Invalid read of size 8" in `refc_binary_decrement_refcount` while running estdlib tests

bettio, 2025-03-19:
Valgrind detected the following issue while running `test_estdlib.avm` with [build-and-test (gcc-11, -O3, 27)](https://github.com/atomvm/AtomVM/actions/runs/13956077909/job/39067390304?pr=1587#logs).

```
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=false Mode=binary Backendoption=[{inet_backend,socket}]
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=false Mode=list Backendoption=[{inet_backend,inet}]
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=false Mode=list Backendoption=[{inet_backend,socket}]
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=true Mode=binary Backendoption=[{inet_backend,inet}]
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=true Mode=binary Backendoption=[{inet_backend,socket}]
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=true Mode=list Backendoption=[{inet_backend,inet}]
GEN_UDP-TEST> SpawnControllingProcess=true IsActive=true Mode=list Backendoption=[{inet_backend,socket}]
+GEN_TCP-TEST> ServerBackendOption=[{inet_backend,socket}] ClientBackendOption=[{inet_backend,socket}] ListenMode=binary ConnectMode=binary ListenActive=false ConnectActive=false
GEN_TCP-TEST> ServerBackendOption=[{inet_backend,socket}] ClientBackendOption=[{inet_backend,socket}] ListenMode=binary ConnectMode=binary ListenActive=false ConnectActive=true
==14935== Thread 4:
==14935== Invalid read of size 8
==14935==    at 0x1A324A: refc_binary_decrement_refcount (refc_binary.c:68)
==14935==    by 0x17404D: memory_sweep_mso_list (memory.c:937)
==14935==    by 0x170399: memory_destroy_heap (memory.h:373)
==14935==    by 0x170399: context_destroy (context.c:195)
==14935==    by 0x119BD7: scheduler_entry_point (opcodesswitch.h:7178)
==14935==    by 0x1B4FE5: scheduler_thread_entry_point (smp.c:75)
==14935==    by 0x4AB4AC2: start_thread (pthread_create.c:442)
==14935==    by 0x4B45A03: clone (clone.S:100)
==14935==  Address 0x8fa6620 is 16 bytes inside a block of size 88 free'd
==14935==    at 0x484B27F: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14935==    by 0x1A330D: refc_binary_destroy (refc_binary.c:91)
==14935==    by 0x1A330D: refc_binary_decrement_refcount (refc_binary.c:70)
==14935==    by 0x170320: context_destroy (context.c:184)
==14935==    by 0x119BD7: scheduler_entry_point (opcodesswitch.h:7178)
==14935==    by 0x1B4FE5: scheduler_thread_entry_point (smp.c:75)
==14935==    by 0x4AB4AC2: start_thread (pthread_create.c:442)
==14935==    by 0x4B45A03: clone (clone.S:100)
==14935==  Block was alloc'd at
==14935==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==14935==    by 0x1A31D8: refc_binary_create_resource (refc_binary.c:39)
==14935==    by 0x1A36A6: enif_alloc_resource (resources.c:68)
==14935==    by 0x1B7484: nif_socket_open (otp_socket.c:523)
==14935==    by 0x1552CB: scheduler_entry_point (opcodesswitch.h:2019)
==14935==    by 0x1B4FE5: scheduler_thread_entry_point (smp.c:75)
==14935==    by 0x4AB4AC2: start_thread (pthread_create.c:442)
==14935==    by 0x4B45A03: clone (clone.S:100)
==14935== 
GEN_TCP-TEST> ServerBackendOption=[{inet_backend,socket}] ClientBackendOption=[{inet_backend,socket}] ListenMode=binary ConnectMode=binary ListenActive=true ConnectActive=false
GEN_TCP-TEST> ServerBackendOption=[{inet_backend,socket}] ClientBackendOption=[{inet_backend,socket}] ListenMode=binary ConnectMode=binary ListenActive=true ConnectActive=true
GEN_TCP-TEST> ServerBackendOption=[{inet_backend,socket}] ClientBackendOption=[{inet_backend,socket}] ListenMode=binary ConnectMode=list ListenActive=false ConnectActive=false
```

UncleGrumpy, 2025-03-19:
This looks like a duplicate of the use after free encountered in the gen_tcp_test #1584. It seems to also be related to #1571.

bettio, 2025-03-24:
> This looks like a duplicate of the use after free encountered in the gen_tcp_test [#1584](https://github.com/atomvm/AtomVM/issues/1584). It seems to also be related to [#1571](https://github.com/atomvm/AtomVM/issues/1571).

Not really, #1571 was related to binaries with OTP-21 compiled code, quite a different issue.

UncleGrumpy, 2025-03-24:
> Not really, [#1571](https://github.com/atomvm/AtomVM/issues/1571) was related to binaries with OTP-21 compiled code, quite a different issue.

Indeed. Paul explained this to me on Telegram. Debugging CI errors can be confusing at times.

commit d416d90c (2025-03-29): Merge pull request #1603 from pguyot/w13/fix-resource-monitor-concurrency-issue

Use ref count and lock on list of monitors for a resource type to ensure that
resource still exists when calling the monitor handler.

Should fix #1589

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #1603 (2025-03-29): Fix concurrency bug related to resource monitors
Use ref count and lock on list of monitors for a resource type to ensure that resource still exists when calling the monitor handler.

Should fix #1589

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2025-03-29 by bettio