atomvm/AtomVM #1590: how to get rid of alisp?

elsbiet, 2025-03-20:
because i am afraid i will soon run out of memory on my esp32 i would like to know how i can remove alisp (i don't need it).

petermm, 2025-03-21:
I think you do it here: https://github.com/atomvm/AtomVM/blob/8eae9576c7a010ed6ad77cbc48411dabbba0818a/libs/CMakeLists.txt#L31

I doubt removing it has much impact on memory, but let us know..

elsbiet, 2025-03-21:
after setting

`set(ATOMVM_LIBS eavmlib estdlib)`

and even after 

`idf.py clean `

the alisp code did not disappear.

did i miss anything?

petermm, 2025-03-21:
atomvmlib is built in the AtomVM/build folder see https://www.atomvm.net/doc/main/build-instructions.html#generic-unix-build-instructions

bettio, 2025-03-23:
Are you worried about available RAM memory?

elsbiet, 2025-03-23:
@peterman: finally i managed to get rid of alisp (it was kind of timeconsuming and painful because the docu seems to be somewhat outdated/inconsistent). removing alisp saved appr 12k which is 5.3%.

@bettio: i'm worried about both ram and flash. i know there is an esp32-s3 with plenty of ram/flash but i'm not sure if and how this extended resources are supported by atomvm.

UncleGrumpy, 2025-03-23:
Removing alisp only saves space in the library partition, so unless you are also changing the application partition beginning this will not give you any more available space for your application. If you application is already too big for the default 1M application partition there is actually unclaimed flash space available at the end of the flash. You can simply remove the size parameter from the `main` partition In partitions.csv (since it is last in the partition table) and it will give you all of the available space for your application.

elsbiet, 2025-03-23:
@UncleGrumpy: thank you very much for your explanation.Things become clearer.

another question: it seems to be usefull (reduced upload time) for me to put parts of my application into boot.avm/lib.avm (btw: is it possible to rename the partitions at will or can this lead to problems?). i assume that it might be necessary to enlarge the partition boot.avm and adjust the offset of main.avm. in addition, the offset in
{atomvm_rebar3_plugin, [
                        `{esp32_flash, [
                                                   {offset, NewOffset}
                                                   ..`
must be adjusted. 

Is there anything else to consider?

pguyot, 2025-03-23:
You can have only two partitions (in addition to nvs and phy_init): one for the VM and one for Erlang/Elixir/Gleam code.

In this case, the AVM you flash on the second partition must include libs such as eavmlib and estdlib. This is typically done with `packbeam -p`. `packbeam` can also prune unreferenced modules. Use this with caution as a module only referenced with `list_to_atom` will be pruned.

This is what is done with La Machine to use as much flash as possible. Likewise, I trimmed down VM by disabling drivers I don't need.

https://github.com/pguyot/la_machine/blob/2b334a4b1a3703a04adb952100901a044d1ce6c3/.github/workflows/build.yaml#L91

As mentioned by @UncleGrumpy you need to define partition.csv accordingly. See for example La Machine as well:

https://github.com/pguyot/la_machine/blob/main/image/partitions.csv

pguyot, 2025-03-24:
Sorry for the confusion.
By default there are 3+2 partitions as you describe.

However, it is possible to have only 2+2.

In the latter case, you can optimize available flash size by:
- only having libs you use
- pruning modules you don't use
- taking advantage of the free space at the end of boot partition

elsbiet, 2025-03-24:
@pguyot thanks for clarification.