atomvm/AtomVM #1591: term_to_binary(Term)  fails if Term is a list of processes.

elsbiet, 2025-03-21:
ProcessesBinary = term_to_binary(erlang:processes())

results in

AtomVM init.
I (858) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting application...
Unknown external term type: 19

abort() was called at PC 0x40156e4b on core 0
0x40156e4b: serialize_term at /home/splge/ATOM/dist/0.7.0-dev+git.025f33bb/src/libAtomVM/externalterm.c:395

Backtrace: 0x40081e6e:0x3ffbb500 0x4008ced1:0x3ffbb520 0x40095d81:0x3ffbb540 0x40156e4b:0x3ffbb5b0 0x40156cbd:0x3ffbb5f0 0x40156e59:0x3ffbb630 0x40156e67:0x3ffbb650 0x40156f5a:0x3ffbb670 0x4014c145:0x3ffbb6a0 0x400f4244:0x3ffbb6c0 0x4012589d:0x3ffbb770 0x400dc7b0:0x3ffbb790 0x402005d3:0x3ffbb7d0 0x4008daf5:0x3ffbb800
0x40081e6e: panic_abort at /home/splge/esp/esp-idf/components/esp_system/panic.c:472
..

petermm, 2025-03-21:
can confirm identical error `:erlang.term_to_binary(self())` on 0.6.6..

pguyot, 2025-03-21:
This is fixed in disterl branch.
See #1358

elsbiet, 2025-03-21:
excuse me, but i'm not very used to github and cannot find the 'disterl branch'. i only see

release-0.6 and feature/distributed-erlang

on 

https://github.com/atomvm/AtomVM/branches/

have you got any suggestions?

petermm, 2025-03-21:
it's the feature/distributed-erlang - disterl is short for distributed erlang..

elsbiet, 2025-03-22:
works fine and binary_to_term es well.

`Processes = erlang:processes(), Processes = binary_to_term(term_to_binary(erlang:processes())).
ok'

btw: i have implemented a  little function for the esp32  which retrieves the unique flash id. (and used this opportunity to simplify the function platform_nifs_get_nif(const char* const nifname) in src/platforms/esp32/components/avm_sys/platform_nifs.c a little.)

if you are interested  in the function and my modifications i would be glad to contribute both,