atomvm/AtomVM #1599: unable to add a custom nif

elsbiet, 2025-03-25:
tried to follow [Adding a custom AtomVM Nif](https://www.atomvm.net/doc/main/build-instructions.html) but 
although i have a clean build there is no file main/component_nifs.txt in src/platforms/esp32. so i have no idea where and how to add my nif name. i created a file main/component_nifs.txt and wrote my nif name into this file but this had no effect.

i cannot find a file named component_nifs.txt file in the entire source tree. so i feel the doc might be kind of outdated.                                                                                                                                                       

is there anybody to tell me how to add my nif?

UncleGrumpy, 2025-03-25:
Indeed. That section of the documentation is outdated. You no longer need to add the nifs or ports to an external file. As long as your nif uses the REGISTER_NIF_COLLECTION macro and the source directly is in the esp32/components directory that should be sufficient for the nifs to be picked up and added to the VM.

UncleGrumpy, 2025-03-26:
Glad to hear you were able to figure it out. If you don’t mind, could you please change the title of this issue to something along the lines of “Documentation for including nifs is out of date”? You can leave it open, this will help remind us to update that section of the docs.

petermm, 2025-03-27:
For the docs/future reference: A user also figured out pico Nifs here: https://erlangforums.com/t/compile-while-calling-own-nifs-error-following-modules-or-functions-are-not-available-on-atomvm-elixir-mymod-mynif-1/4502 code: https://github.com/mx-ws/AtomVM/tree/pico-pio

fwiw I do believe esp32cam is outdated pending a PR merge, and maybe https://github.com/atomvm/atomvm_neopixel is a better example at the moment..

UncleGrumpy, 2025-03-27:
> For the docs/future reference: A user also figured out pico Nifs here: [...]

Thank you! I knew there was a better example, I was just having a herd time remembering which use nifs vs ports or pure Erlang drivers and went with the first one I found ;)

commit 02c52740 (2025-05-13): Merge pull request #1659 from UncleGrumpy/docs_build_update

Update build documentation for custom nif or port inclusion

Closes #1599

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit 600fdcdb (2025-05-13): Update build documentation for nif or port inclusion

Updates "Build Instructions"  documentation for including custom nifs and ports into esp32 builds.

Closes #1599