atomvm/AtomVM #1607: Is I2C supported on platforms pico / RP2040?

gpummer, 2025-03-30:
I tried the bme280 sample (esp32) on my pico, but had crashes on i2c_bus: and/or i2c:open/1. I searched the issues and come to the conclusion that I2C is not supported at the moment. Is this correct?

If so ...  I could try to implement the required driver/nif. Is gpiodriver.c (from src/platforms/rp2/src/lib) or i2c_driver.c (from src/platforms/esp32/components/avm_builtins) a good starting point? Or even better, has someone already done this? I'm not an expert, neither for esp32 nor pico, erlang or AtomVM, but I speak C and some other programming languages. And yes, I know that i2c has many different use cases.

Oh! I would like to hijack this issue, to thank all the people working on AtomVM! You do great work! There is so much documentation! The VM is stable and there is so much library support! It is a pleasure to have a software platform that supports multiple hardware targets. Many thanks!

pguyot, 2025-03-30:
Indeed i2c is not implemented on the rp2 platform.

This is somewhat in my backlog but definitely not at the top. A PR would be more than welcome.

ESP32 driver is a good starting point.

As general guidance:

- you will find out it has many esp-isms. We have yet to define a clean hardware abstraction layer. I would personally prefer nifs that match the underlying sdk and Erlang glue on top.

- we try to favor nifs and resources over ports. This is especially true with rp2 that doesn't have threads/tasks

gpummer, 2025-03-31:
Many thanks for the infos! I appreciate the guidance and will try my best.