atomvm/AtomVM #1640: esp32:i2c - confusion

elsbiet, 2025-04-29:
hello,

there are 2 interfaces to the i2c-bus in avm_builtins:

1. portbased: components/avm_builtins/i2c_driver.c
2. nifbased: components/avm_builtins/i2c_resource.c

but why do atomvm_ssd1306.c, ssd1306_display_driver.c and ds3231.c implement their own interfaces?

moreover i feel that the sda and the scl pin definitions are properties of the hardware design and not properties of any peripheral device such as ssd1306 or ds3231.c and possibly others. for this reason i think the definitions of the sda and the scl pin should be done in  i2c_driver.c : i2c_driver_init(..) / i2c_resource.c  : i2c_resource_init respectively and not in any device implementations. 

the configuration of the 2 pins could be done by means of idf.py menuconfig.

one advantage would be that multiple i2c devices can be accessed without error  prone multiple initialization  

what do you think?

petermm, 2025-04-30:
Agreed that some streamlining is needed..

Believe avm_builtins/i2c_driver.c and avm_builtins/i2c_resource.c are direct drivers, port/nif with pro/cons, which is fine, though only having one, would be simpler.

Additionally there is i2c_bus https://github.com/atomvm/atomvm_lib/blob/master/src/i2c_bus.erl for interfacing multiple i2c devices on top of a single port/nif, and you have https://github.com/arpunk/atomvm_ssd1306/tree/erlang_driver for using ssd1306 over the i2c_bus.erl - but it's not currently at feature parity with the native driver? see https://github.com/atomvm/atomvm_ssd1306/pull/3.

the https://github.com/atomvm/atomvm_ssd1306 project currently have a native driver and https://github.com/atomvm/atomgl as well.

To further muddy the waters, I believe all of these are using the deprecated V1 i2c api - https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c_slave_v1.html#i2c-slave-v1 - so we should update them to V2 api, but this is perhaps a separate/tangent issue.

I believe the "winners" are the https://github.com/arpunk/atomvm_ssd1306/tree/erlang_driver for ease of installation (being pure erlang), for the simple stuff, and then atomgl for features, and native performance..

But I certainly don't have a full understanding of the display drivers..

UncleGrumpy, 2025-04-30:
The nif based implementation is new (still unreleased) and still considered experimental.

I believe the standalone ssd1306 driver predates the general purpose i2c driver.

The reason pins cannot be configured with idf.py menuconfig is that most users are expected to be using download release images, making configuration with idf.py impossible.