atomvm/AtomVM #1647: esp32-s3:  problem creating an returning a list from a nif

elsbiet, 2025-05-05:
i just ran into a problem when creating an returning a list from in a nif.

nif_code:

static term nif_test_cr8list(Context *ctx, int argc, term argv[]) {
    term result_list = term_nil();
    for (int i = 29 - 1;  i >= 0;  --i) {
        result_list = term_list_prepend(term_from_int(i), result_list, &ctx->heap);
    }
    term result = term_alloc_tuple(2, &ctx->heap);
    term_put_tuple_element(result, 0, OK_ATOM);
    term_put_tuple_element(result, 1, result_list);
    return result;
}

caller code:

-module(main).

-include_lib("kernel/include/logger.hrl").

-export([start/0]).

start()  ->
    {ok, _Pid} = logger_manager:start_link(#{log_level => debug}),
    ?LOG_DEBUG("logger started"),

    {ok, L} = test:cr8list(),
    lists:foreach(fun(E) -> io:format("~2.16.0B ", [E]) end, L),
    io:format("]~n")
    .

result:
###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ## 
      ## ##      ##    ##     ## ###   ### ##     ## ###   ### 
     ##   ##     ##    ##     ## #### #### ##     ## #### #### 
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ## 
    #########    ##    ##     ## ##     ##  ##   ##  ##     ## 
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ## 
    ##     ##    ##     #######  ##     ##    ###    ##     ## 

    ###########################################################

I (579) AtomVM: Starting AtomVM revision 0.7.0-dev
I (589) sys: Loaded BEAM partition boot.avm at address 0x1d0000 (size=655360 bytes)
I (609) network_driver: Initialized network interface
I (609) network_driver: Created default event loop
I (619) AtomVM: Found startup beam esp32init.beam
I (619) AtomVM: Starting esp32init.beam...
---
AtomVM init.
I (619) sys: Loaded BEAM partition main.avm at address 0x270000 (size=1048576 bytes)
Starting application...
1970-01-01T00:00:01.154Z [debug] <0.1.0> main:start/0 (/home/splge/ATOM/MyAtomVM/erlang/esp32/i2ct/src/main.erl:9) logger started
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C ]
Undecoded opcode: 34
failed at 7

abort() was called at PC 0x420573d7 on core 0
0x420573d7: scheduler_entry_point at /home/splge/ATOM/dist/AtomVM-feature-distributed-erlang/src/libAtomVM/opcodesswitch.h:7269
..

when i replace 29 in nif_test_cr8list by 28 than the program terminates the program hangs after printing the result list.

when i replace 29 by 27 the program terminates as expected 
``..
AtomVM finished with return value: ok
I (834) AtomVM: AtomVM application terminated.  Going to sleep forever ...
..

is there anybody who can tell me what i might have missed?

thanks for any answer.

elsbiet, 2025-05-05:
As a temporary workaround, I created a binary that contains all the return list items and created a list from this binary on the caller's side.

pguyot, 2025-05-08:
It seems your nif isn't allocating memory for your list. You need to call one of memory_ensure_free functions with TUPLE_SIZE(2) + LIST_SIZE(29) or something similar.

elsbiet, 2025-05-10:
tahnk yuo very much for your answer.

changed nif to
`static term nif_test_cr8list(Context *ctx, int argc, term argv[]) {
    
    int max_count = 29;
    
    if (UNLIKELY(memory_ensure_free(ctx, TUPLE_SIZE(2) + LIST_SIZE(max_count, sizeof(int))) != MEMORY_GC_OK)) {
        ESP_LOGW(TAG, "Failed to allocate memory: %s:%i.\n", __FILE__, __LINE__);
        return ERROR_ATOM;
    }   
    term result_list = term_nil();
    for (int i = max_count;  i >= 0;  --i) {
        result_list = term_list_prepend(term_from_int(i), result_list, &ctx->heap);
    }   
    term result = term_alloc_tuple(2, &ctx->heap);
    term_put_tuple_element(result, 0, OK_ATOM);
    term_put_tuple_element(result, 1, result_list);
    return result;
}
`
and it works fine now.

are there any hints on how to implement a nif, especially how to handle the memory?

jgonet, 2025-06-09:
@elsbiet 
The easiest way is to read some nifs. `nif_erlang_tuple_to_list_1` is some simple example. Context argument contains `heap` field which is heap memory managed by AtomVM. A big simplification is to think of it as an array with some capacity and an index to last allocated object (list, tuple, etc).

If you want to create a new tuple, you need to make sure that this array has enough space. You do it by calling `memory_ensure_free_...` family of functions. Each object has a size, in terms (either 4 or 8 bytes). Here, in `nif_erlang_tuple_to_list_1` we have `tuple_size * 2`, because we want a list with every element in a tuple (*reference* to tuple is one term – every element already lives in heap, and we need 2 terms for every list cell – search for CONS).

Roots are an array of terms that you want to keep reference to in case heap changes and rewrites memory addresses. You pass existing reference to term and it overwrites with new value when GC runs.

In your case, you allocate space for 2-tuple and a list of terms (each taking one term). The correct space calculation is `TUPLE_SIZE(2) + LIST_SIZE(max_count, 1)`. You don't specify bytes, you specify terms.

Feel free to close the issue if it answered your questions.